esp-rs/esp-hal #2083: Support XiP from PSRAM

Dominaezzz, 2024-09-04:
https://docs.espressif.com/projects/esp-idf/en/stable/esp32s3/api-guides/external-ram.html#execute-in-place-xip-from-psram

[CONFIG_SPIRAM_FETCH_INSTRUCTIONS](https://docs.espressif.com/projects/esp-idf/en/stable/esp32s3/api-reference/kconfig.html#config-spiram-fetch-instructions) and [CONFIG_SPIRAM_RODATA](https://docs.espressif.com/projects/esp-idf/en/stable/esp32s3/api-reference/kconfig.html#config-spiram-rodata) are the specific esp-idf configs.

In short, this feature lets you copy the application and data from flash to PSRAM at start up, which means flash is no longer needed for the rest of your application.

This is important for #2081 where you want to store huge buffers in PSRAM and DMA them to a peripheral, but you don't want flash access temporarily disabling psram access, leading to the DMA being starved of data and the peripheral getting garbage when it inevitably goes too far ahead.

Somewhat related #1083 .

bjoernQ, 2024-09-17:
Turns out at least on ESP32-S3 we can execute code from PSRAM right after initializing PSRAM as we currently do - it's just that it's not writable via ibus (but we can via dbus and need to make sure to synchonize caches)

I think there are two ways

- the original request from #1083 extended by linking specific functions into PSRAM - for this we would need to create suitable sections in the linker scripts, copy code and data to PSRAM before anyone is allowed to access it. One challenge might be that currently we map PSRAM right after FLASH so the linker script needs to calculate the start address accordingly
- the way ESP-IDF does it (i.e. copy everything to PSRAM and then map PSRAM at the origin of external address space) - see https://github.com/espressif/esp-idf/blob/3c99557eeea4e0945e77aabac672fbef52294d54/components/esp_psram/mmu_psram_flash.c#L46-L134

yanshay, 2024-12-23:
I'd like to place my vote in favor of implementing this issue and explain the rationale.

If I understand this right, this feature is practically required for applications that need to use large RGB displays (where buffer is in PSRAM) while at the same time using PSRAM intensively for application use.

Well, for complex applications, with user interface, using WiFi, SSL, async, etc. the limited amount of RAM on esp32 is quickly exhausted and PSRAM is required for memory allocations.
And for large displays (over 3.5") RGB is mandatory and required PSRAM memory for buffering.

There are many applications of this sort out there, and currently they are all written in C++ and out of the reach of rust developers.

Therefore, to enter the space of UI based applications, this feature is a must. I think that would be an effort wisely invested.

And to sum up: 
_"That's one small step for esp-hal, one giant leap for rustaceans."_

bjoernQ, 2025-01-16:
The simplest thing to try I can think of would be:

- map PSRAM as we already do, FLASH is still mapped
- just memcopy everything from the mapped flash to psram
- map psram to the origin of extram, invalidate caches

I _think_ ESP-IDF directly reads flash / writes psram via SPI but just memcpy the mapped flash/psram should be easier (no code to port from ESP-IDF)

I haven't tried that (since then we would already have it :) ) but I guess it could work this way. All necessary changes would be in https://github.com/esp-rs/esp-hal/blob/4b66e3dba736f30ca370f670a8d3e1012fde50a3/esp-hal/src/soc/esp32s3/psram.rs#L116-L234

EliteTK, 2025-01-19:
The recent 38C3 talk on liberating the WiFi on the ESP32 mentioned that reverse-engineering the ROM is allowed by espressif. So, after a bit of reverse engineering work and studying the IDF sources I've gotten XiP from PSRAM working on my ESP32S3.

<https://gist.github.com/EliteTK/5a409431082b4a4c34bb560243f2cf61>

The code is ported from the equivalent ESP-IDF with any excess fluff removed. The comments contain the reverse engineered, cleaned up, mostly valid C which represents what the `Cache_` functions which get called are actually doing.

I haven't actually tested this with a display yet as I never tried writing the DMA from PSRAM code mentioned in the related issues because I knew (from experience with esp-idf-hal, before I enabled XiP from PSRAM) that it would be glitchy and unusable. But I hope to do that tomorrow or next week.

I think it would make sense to put this in `init_psram` with additional PsramConfig fields for specifying that you want this feature. Putting the .text and .rodata in PSRAM has an effect on the amount of remaining available PSRAM which is another reason why I think it make sense to do this there. I plan on putting together a draft PR to this effect.

The code for doing this for the ESP32-S2 and ESP32-S3 is very similar. There's a v2 file in the IDF which handles ESP32-P4. I don't actually know enough about ESP32 to know if that's the extent of SOCs which support this feature.

In general I have minimal experience with esp-hal, limited experience with embedded rust, limited experience with unsafe rust, and only about a year of proper rust experience so please do comment on the gist on anything you think looks odd/could be improved, I'll happily incorporate it as I keep working on this.

bjoernQ, 2025-01-20:
Wow - didn't notice there is a `Cache_Flash_To_SPIRAM_Copy` ROM function - nice

EliteTK, 2025-01-23:
I've now opened a WIP PR for this in #3024 . I would appreciate any and all feedback.

latesr, 2026-10-08:
I have implemented most of this and tested it on the ESP32-C5 and the ESP32-S31. I was hoping to agree the shape of pull requests that would work for the project. My implementation builds on the ideas in this issue, in #3024 and in #1090.

My ultimate need and aim is to keep code running from PSRAM with the cache on and without stalls through any flash erase or write (see PR 7 below).

## Proposal

1. An esp-config option, `code-in-psram`, off by default and unstable. It is a build-time option, because the linker script will change on chips with a separate PSRAM MMU. #3024 adds a runtime field to `PsramConfig`, which suits the S3. However, that doesn't work for a chip with a separate PSRAM MMU that needs a different linker script. Chips that support the option would get a new PSRAM metadata property, `code_in_psram`.
2. With the option on, esp-hal starts PSRAM from its esp-config settings, as bjoernQ noted in #1090. On the C5 this happens in `esp_hal::init`, and on the S31 before `esp_hal::init`. The only setting so far is `psram-frequency`. `Psram::new` and `psram_allocator!` keep working, and give the heap the pages _after_ the copied code and read-only data. A `Psram::new` with a different `PsramConfig` warns and keeps PSRAM as it is, because PSRAM config can't be changed whilst running from it.
3. On the C5, with one MMU for flash and PSRAM, the link doesn't need to change. Each page that the flash MMU maps is copied to a PSRAM page, and its entry in the MMU table is rewritten to point at PSRAM at the same virtual address. This is the approach sketched earlier in this issue, and the one ESP-IDF follows in `mmu_psram_flash_v2.c`. The copy and the remap run from RAM. The bootloader entry for the app's first page stays as is. OTA requires the `booted_partition` fix, PR 1 below, which stops it reading entry 0.
4. On the S31, with a separate PSRAM MMU, `.text` and `.rodata` are linked in the PSRAM window at `0x5000_0020`. This follows ESP-IDF's `memory.ld.in` with `SPIRAM_FETCH_INSTRUCTIONS` and `SPIRAM_RODATA`. ESP-IDF's bootloader skips segments in the PSRAM window and keeps one flash entry at the app's first page. esp-hal starts PSRAM and copies the segments from flash in `riscv_preinit`, after `.bss` is zeroed and before anything reads external memory. ESP-IDF's `cpu_start.c` does the copy at the same point. It finds segments as ESP-IDF's `image_process.c` does. With the option on, riscv-rt's start-up sections link into RAM, to ensure everything on that path runs from RAM, including the MPLL and PSRAM start-up. espflash will also need to place segments in the PSRAM window in the image, and I have a small change for that. The C5 does not need an espflash change. The P4 has the S31's shape, with a separate PSRAM MMU, so it could follow the same path. I have no P4 board to test it on.
5. One `mmu` module shared by `flash` and `psram`, with the per-chip register access behind `cfg_select!` as in `flash/mmu.rs`. This is based on the review comments in #3024.
6. With flash encryption on, the copy is plaintext until esp-hal supports encrypted PSRAM. Until then, I'd propose esp-hal warns in this configuration and esp-config has a clear explanation. `map_psram` does not encrypt PSRAM today, so a heap in PSRAM is already plaintext.

`esp_hal::flash` suspends the caches during a flash operation, so code in PSRAM still stops for a write or an erase.

## Proposed pull requests

I've tried to split it up so it's reviewable. The first four stand on their own. PR 5 needs PRs 1 and 2. PR 6 needs PRs 3, 4 and 5, and the espflash and probe-rs releases. PR 7 needs PR 5.

1. `booted_partition` and the running-app check in `partitions/region.rs` find the app from the flash entry the bootloader maps for it, at the top of the MMU table, and read the full page field. Today they read entry 0, and the C5, C6, C61 and H2 mask it to 8 bits.
2. `Psram::size`, which returns the detected size, as ESP-IDF's `esp_psram_get_size` does. With code in PSRAM, `raw_parts` returns less than the chip holds. It also fixes the ESP32's "PSRAM size" log, which prints 0 MB with the default auto-detect, even on boards that have PSRAM.
3. The S31's MPLL at 400 MHz as well as 500 MHz. The ESP32-S31-WROOM-3 rates its PSRAM at 200 MHz, and esp-hal's 200 MHz preset won't work on a 500 MHz MPLL. It adds 400 MHz without changing any default.
4. The cache write-back on the S31 and the P4 issues its sync twice, as ESP-IDF's ROM patch does "to avoid sync loss" (`ESP_ROM_CACHE_WRITEBACK_NEEDS_SYNC_TWICE_MAP`). esp-hal uses the write-back for DMA buffers today, and the S31 copy needs it.
5. Code in PSRAM on the C5. It includes a commit that moves the MMU entry helpers into a shared `mmu.rs`. With the option on, the `flash` HIL test `test_mapped_drom_read_after_program` checks that a mapped `.rodata` read stays as it was after a write, while `Flash::read` returns the new data.
6. Code in PSRAM on the S31. It includes a commit that makes the PSRAM start-up safe to run from RAM. Its HIL test needs an espflash release with the PSRAM window change, which I'll send a PR for, and a probe-rs release that takes it.
7. Flash operations keep the caches on when no code runs from flash on both chips. I recognise this touches #6207 and would need some discussion about approach.

## Questions

[... cut ...]  run on an S31 build with the option on, such as the `code_in_psram` HIL test, that disassembles the ELF and fails on any reference from the RAM sections into the PSRAM window? It would use `llvm-objdump` from the `llvm-tools` component, which the binary-size workflow already installs, much as `check-global-symbols` inspects build output today.
- Does the split above suit you, or would you prefer fewer, larger pull requests?
- @EliteTK, would the S3 in #3024 fit the same option and `mmu` module?

bugadani, 2026-10-08:
@latesr your wall of text makes me think (personally, IDK the team's stance just yet) this needs to be done carefully.

- We should research which chips allow what, and come up with a plan first. Information at this point is more important than an initial implementation. Bandwidth issues with DMA and PSRAM buffers is one of my concerns. "a separate PSRAM MMU that needs a different linker script." is also something I'd like to see explained - ideally not by an LLM-written post, but actually something a human understands.
- Once we have the information, we can formulate a plan that will work for all chips, if that is possible, or classify chips into support shapes.
- Chip support can be introduced one by one. I have some trouble separating C5 and S31 in your comment, and they don't need to be done at the same time. At the very least, one tracking issue per shape could help discussions and design.

latesr, 2026-10-08:
Hi @bugadani.  Apologies for the long comment. I wasn't sure of the best way to communicate it all as the initial PRs stand alone, but make more sense given the wider goal. I am very willing to work with you and others to split this up in whatever works best for the project. 

Regarding the "separate PSRAM MMU", here's what I meant.

On the C5 there's one MMU table. Each entry maps a 64 KB page in the `0x4200_0000` window, and a bit in the entry says whether the page comes from flash or PSRAM. So you can copy a page of code to PSRAM, point the MMU entry at the copy, and the code keeps running at the same address. The linker script doesn't change.

The S31 is different. Flash and PSRAM have their own windows, `0x4000_0000` and `0x5000_0000`, and each has its own MMU table in a different SPI memory controller. Flash and PSRAM need to be mapped separately in different tables and at different windows. So code can't move to PSRAM and keep its flash address. It has to be linked at a `0x5000_0000` address, which is what ESP-IDF's memory.ld.in does with `SPIRAM_FETCH_INSTRUCTIONS`. That requires a different linker script.

I can research what the other similar chips do and help start a categorisation of them if that's useful. My intent with the above was that each chip would opt into it with a `code_in_psram` metadata property.

Could you elaborate a bit about the DMA and PSRAM bandwidth concern? Are you concerned about contention between code fetches and DMA peripheral buffers in PSRAM, or a limit on particular chips?

latesr, 2026-10-08:
Regarding the research request, I've looked at how each chip with PSRAM could run code from it, using ESP-IDF master and esp-hal main. They fall into three groups. 

| Group | Chips | How code gets into PSRAM | Different linker script required |
|---|---|---|---|
| 1. Not possible | ESP32 | PSRAM is only mapped as data, no execution | N/A |
| 2. One MMU for flash and PSRAM | S2, S3, C5, C61, H4 | Copy each mapped page to PSRAM and point its MMU entry at the copy, at the same address | No |
| 3. A separate flash and PSRAM MMU / window | S31, P4 | Link `.text`, `.rodata`, etc in the PSRAM window and copy them from flash at boot | Yes |

I've validated the C5 and S31 rows on my boards. All the others are based on ESP-IDF code and documentation. 

ESP-IDF has a flag indicating it sees a similar split between group 2 and 3: `SOC_MMU_PER_EXT_MEM_TARGET` in `components/soc/<chip>/include/soc/soc_caps.h`.

Additional details:

Group 2: the S3 has one MMU table for its instruction and data windows, so it works like the C5. #3024 does this on the S3 at run time, from `esp_hal::init`. The S2 follows the same idea, but its MMU has one table per bus, so each bus's entries are remapped separately.

Group 3: The S31 PAC has a PSRAM bit in the MMU entry (bit 10). I have double checked on my S31 and whilst you can set and retrieve that bit, the S31 doesn't map to PSRAM when you do. So a group 3 link change is required. Additionally, as PSRAM settings like frequency are required before `main`, this needs to be a build-time option that seems best to handle in esp-config. 

Regarding DMA: every chip except the ESP32 can DMA to and from PSRAM. In group 2, PSRAM shares the SPI bus with flash, so code fetches use the same bus as before. In group 3, PSRAM has its own, much faster bus (octal DDR PSRAM at 200 MHz, 400 MB/s, against quad flash at 80 MHz, 40 MB/s, on the S31-WROOM-3), so code fetches add to its load. Whether that helps or hinders overall will depend on load and the specific use case.

In my use case, I need deterministic, low thread latency even during flash erase and write. Today every thread waits for the whole erase, up to 500 ms for a 4 KB sector on the S31-WROOM-3. With code in PSRAM, nothing needs flash while it erases, so flash operations can keep the caches on and the other core running, and that whole class of latency problems goes away.

I recognise others will prioritise different concerns. Display and camera frame buffers are one example: they're often too large for SRAM, so their DMA has to use PSRAM, and code in PSRAM would compete with it.

The variety of needs is the reason I've proposed it as opt-in at esp-config and only for the chips that have declared support for it in their metadata. My intuition remains that there's benefit in a common esp-config approach to this feature across all of group 2 and 3 chips that have opted in. However, if you'd prefer group 2 to be configured at run time in application code, I'm happy to drop the common esp-config approach and only implement the S31, which needs build-time configuration either way.

blowfishpro, 2026-10-09:
> I recognise others will prioritise different concerns. Display and camera frame buffers are one example: they're often too large for SRAM, so their DMA has to use PSRAM, and code in PSRAM would compete with it.

So it turns out that moving code to PSRAM can actually be desirable if not necessary for many cases, see [this comment]
(https://github.com/esp-rs/esp-hal/issues/2884#issuecomment-3841221242) for an explanation of why.