esp-rs/esp-hal #2089: ncorrect CAN Frame ID during Message Transmission

vibhanshuverma02, 2024-09-05:
Here is your formatted GitHub issue text:

Issue: Incorrect CAN Frame ID during Message Transmission

I am writing an application for the ESP32 using the CAN protocol. I have used the esp_hal::Twai crate for this implementation. However, I am facing an issue with CAN message transmission. The byte array data for the CAN message has been encoded into different bits, and I have set the CAN Frame ID as zero according to the provided example. However, it is being encoded into another Frame ID, specifically 0x3B9.

Below is the code I used along with the CAN dump output.
Code:

rust

#![no_std]
#![no_main]

const IS_FIRST_SENDER: bool = true;
use esp_backtrace as _;
use embedded_can::nb::Can;

use esp_hal::{
    clock::ClockControl,
    gpio::Io,
    peripherals::Peripherals,
    prelude::*,
    system::SystemControl,
    twai::{self, filter::SingleStandardFilter, EspTwaiFrame, StandardId},
};
use esp_println::println;
use nb::block;

#[entry]
fn main() -> ! {
    let peripherals = Peripherals::take();
    let system = SystemControl::new(peripherals.SYSTEM);
    let clocks = ClockControl::boot_defaults(system.clock_control).freeze();

    let io = Io::new(peripherals.GPIO, peripherals.IO_MUX);

    let can_tx_pin = io.pins.gpio5;
    let can_rx_pin = io.pins.gpio4;

    // The speed of the CAN bus.
    const CAN_BAUDRATE: twai::BaudRate = twai::BaudRate::B500K;
    println!("baud: {}", match CAN_BAUDRATE {
        twai::BaudRate::B500K => "500K",
        twai::BaudRate::B125K => "125K",
        twai::BaudRate::B250K => "250K",
        twai::BaudRate::B1000K => "1000K",
        twai::BaudRate::Custom(_) => todo!(),
    });
    
    let mut can_config = twai::TwaiConfiguration::new(
        peripherals.TWAI0,
        can_tx_pin,
        can_rx_pin,
        &clocks,
        CAN_BAUDRATE,
        None,
    );

    // Define the CAN filter
    // let filter = SingleStandardFilter::new(StandardId::new(0x000).unwrap(), 0x000);
    // can_config.set_filter(filter);

    let mut can = can_config.start();

    loop {
        // Send the CAN message
        let frame = EspTwaiFrame::new(StandardId::new(0x010).unwrap().into(), &[14, 20, 3]).unwrap();
        println!("Frame data: {:?}", frame);

        match block!(can.transmit(&frame)) {
            Ok(_) => println!("Frame transmitted"),
            Err(e) => println!("Failed to transmit frame: {:?}", e),
        }
    }
}

CAN Dump Output:
[image]

Problem:

    The CAN Frame ID is set to 0x010 in the code, but the actual transmission is showing a different Frame ID (0x3B9).
    I would like to understand how to correctly transmit the actual CAN message without modifying the Frame ID.

Any help or clarification on how to resolve this issue would be greatly appreciated.

playfulFence, 2024-09-30:
Hello!
I tried reproducing your problem (but using SelfTest mode) and for me, if I did everything right, of course everything works as expected.
My updated code just in case: 
```rust
#![no_std]
#![no_main]

const IS_FIRST_SENDER: bool = true;
use esp_backtrace as _;
use embedded_can::nb::Can;

use esp_hal::{
    // clock::ClockControl,
    gpio::Io,
    peripherals::Peripherals,
    prelude::*,
    // system::SystemControl,
    twai::{self, filter::SingleStandardFilter, EspTwaiFrame, StandardId},
};
use esp_println::println;
use nb::block;

#[entry]
fn main() -> ! {
    let peripherals = esp_hal::init(esp_hal::Config::default());

    let io = Io::new(peripherals.GPIO, peripherals.IO_MUX);

    let can_tx_pin = io.pins.gpio5;
    let can_rx_pin = io.pins.gpio4;

    // The speed of the CAN bus.
    const CAN_BAUDRATE: twai::BaudRate = twai::BaudRate::B500K;
    println!("baud: {}", match CAN_BAUDRATE {
        twai::BaudRate::B500K => "500K",
        twai::BaudRate::B125K => "125K",
        twai::BaudRate::B250K => "250K",
        twai::BaudRate::B1000K => "1000K",
        twai::BaudRate::Custom(_) => todo!(),
    });
    
    let mut can_config = twai::TwaiConfiguration::new(
        peripherals.TWAI0,
        can_tx_pin,
        can_rx_pin,
        CAN_BAUDRATE,
        crate::twai::TwaiMode::SelfTest,
    );

    // Define the CAN filter
    // let filter = SingleStandardFilter::new(StandardId::new(0x000).unwrap(), 0x000);
    // can_config.set_filter(filter);

    let mut can = can_config.start();

    loop {
        // Send the CAN message
        let frame = EspTwaiFrame::new(StandardId::new(0x010).unwrap(), &[14, 20, 3]).unwrap();
        println!("Frame data: {:?}", frame);

        match block!(can.transmit(&frame)) {
            Ok(_) => println!("Frame transmitted"),
            Err(e) => println!("Failed to transmit frame: {:?}", e),
        }
    }
}
```

Judging by the way the code in issue is written, you are not using the most up to date version of `esp-hal`. Can I ask you to try using `main` just to make sure that this problem has been fixed?

[image]

bugadani, 2024-09-30:
@playfulFence you can't easily reproduce a TWAI frame that accidentally has 8+ in the DLC field - we don't provide the API to generate such frames and I think it's fairly rare to receive such a frame by accident with a correct CRC.

Technically, we have enough data registers in the hardware, but I'm not sure if we can actually mis-use them to generate an incorrect frame like the one in question here.

playfulFence, 2024-10-29:
@vibhanshuverma02, when you get a chance to test your usecase with the current version of a driver (latest release or `main`) - please let us know if the problem is still present! In case the problem is confirmed - I will immediately start an investigation.

playfulFence, 2024-11-08:
Okay, since there has been no response, we will close this issue.
If you find the same problem again in the latest driver versions, please feel free to open the issue again!

closed 2024-11-08 by playfulFence