esp-rs/esp-hal #2098: ESP32-S2: SPI DMA test_asymmetric_dma_transfer fails

SergioGasquez, 2024-09-06:
For some reason, [S2 HIL](https://github.com/esp-rs/esp-hal/actions/runs/10717649673/job/29718015468) is failing this test, I can also reproduce it locally.  Not sure why it only happens in S2 and when using different buffer sizes (using `dma_buffers!(4,4)` passes the test)

`main` doesnt have any issue with this test in S2

_Originally posted by @SergioGasquez in https://github.com/esp-rs/esp-hal/pull/2074#discussion_r1745182546_

Dominaezzz, 2024-09-06:
Ah yeah it is almost definitely [this](https://github.com/esp-rs/esp-hal/issues/1524#issuecomment-2257254057).

For the S2 the fix is trivial. You just need to change https://github.com/esp-rs/esp-hal/blob/492e35aa74d7b6dfc9d8b5cfc3e11ae6e637378f/esp-hal/src/spi/master.rs#L3459-L3479 to pass the correct lengths for the read and write side rather than the max of both.

Dominaezzz, 2024-09-06:
It's also worth noting that this bug would've been swept under the rug (like it is for some of the other chips) if the pdma was being reset during construction like the Gdma currently is.

bugadani, 2024-09-17:
Dominic, are you sure this isn't a timing issue? The test passes if I enable `defmt` which is ironic because without `defmt` I can't see the backtrace when the test panics :D

Dominaezzz, 2024-09-17:
It may very well be, unfortunately probe-rs doesn't seem happy with my S2 so I can't really dig in further. 

And given the number of changes that have happened since I made that hypothesis, I won't be surprised if I'm wrong haha

bugadani, 2024-09-17:
Thanks for your input, I'm looking into this :)

closed 2024-09-19 by jessebraham