esp-rs/esp-hal #2102: psram.rs raises exception during initialization

clabbenius, 2024-09-06:
Im getting LoadProhibited when trying to initialize psram on esp32s3 causing  exception at Cache_Suspend_DCache() in psram.rs 

Exception occured 'LoadProhibited'
Context
PC=0x42008d0b       PS=0x00060930
0x42008d0b - esp_hal::soc::implementation::psram::init_psram
    at /home/nordmcla/.cargo/registry/src/[index.crates.io](http://index.crates.io/)-6f17d22bba15001f/esp-hal-0.20.1/src/soc/esp32s3/[psram.rs:125](http://psram.rs:125/)
A0=0x82041d02       A1=0x3fcd9d60       A2=0x3fcb2568       A3=0x00000090       A4=0x3fc90d88
0x3fcb2568 - survey::__embassy_main::POOL
    at ??:??
0x3fc90d88 - log::LOGGER.1
    at ??:??
A5=0x00000090       A6=0x00000020       A7=0x00000008       A8=0x82008d09       A9=0x3fcd9d40
A10=0x00004000      A11=0x00000004      A12=0x00000007      A13=0x00000800      A14=0x00004000
A15=0x00010000
SAR=00000020
EXCCAUSE=0x0000001c EXCVADDR=0x00000090
LBEG=0x403cea32     LEND=0x403cea36     LCOUNT=0x00000000
THREADPTR=0x00000000
SCOMPARE1=0x00000100
BR=0x00000000
ACCLO=0x00000000    ACCHI=0x00000000
M0=0x00000000       M1=0x00000000       M2=0x00000000       M3=0x00000000
F64R_LO=0x99a080a1  F64R_HI=0x0020c020  F64S=0xbd910899
FCR=0xc04e0cfe      FSR=0x09a80020
F0=0x2c10aab0       F1=0x11bb303b       F2=0xc020aab0       F3=0x09a90020       F4=0xa80020c0
F5=0xb0eb7c08       F6=0x20c010aa       F7=0xc108a900       F8=0x20c0feb5       F9=0xd00ca800
F10=0x20c020aa      F11=0xc00ca900      F12=0x09a80020      F13=0xb0feb4c1      F14=0x20c010aa
F15=0xc009a900
0x42044726
main

By doing some minor println or such in the code can cause it to start without having any exception. Seems like a timing / compilation optimization thing? 
Saw there was some esp idf config items related to esp32s3 and cache bug, don't know if its related.

bjoernQ, 2024-09-06:
Seems like moving `init_psram` to RAM makes it work reliably

That is contained in the (otherwise unrelated) PR #2099

clabbenius, 2024-09-09:
Okay, will have a go at testing it, do we know when next version of esp-hal will be set?

Also something related, have there been issues with app core running and flash accessing? Got into some lockups when trying to write to flash from main core if app core was running with some task.

bjoernQ, 2024-09-09:
> Also something related, have there been issues with app core running and flash accessing? Got into some lockups when trying to write to flash from main core if app core was running with some task.

It shouldn't since flash access is done in a critical-section and functions should run from internal RAM - if you can create a minimal repro please create an issue for this

MabezDev, 2024-09-09:
> Also something related, have there been issues with app core running and flash accessing? Got into some lockups when trying to write to flash from main core if app core was running with some task.

Yes, the second core should only be running the code it's executing is already in the cache/RAM - any access to flash whilst another core is writing to it will cause problems.

We had a PR for esp-storage: https://github.com/esp-rs/esp-storage/pull/29 which didn't end up getting merged, but we might want to think about revisiting it.

avafloww, 2024-09-15:
While the next version is still in the works, I've backported this fix on top of v0.20.1; just add to your `Cargo.toml`:

```toml
[patch.crates-io]
# backported fix for https://github.com/esp-rs/esp-hal/issues/2102
esp-hal = { git = "https://github.com/avafloww/esp-hal.git", rev = "ebcefe73ee5c6977c631e42d39d8143682478d58" }
```

MabezDev, 2024-09-17:
Closing as this has been resolved in main.