esp-rs/esp-hal #2108: How does sleep work on multicore chips?

dimpolo, 2024-09-06:
I noticed that calling `Rtc::sleep_deep` from the second core does not put the first core to sleep.
Is this expected behavior? Is it possible for the second core to put both cores to sleep?

dimpolo, 2024-09-09:
Actually I was doing something wrong and `Rtc::sleep_deep` works as expected and puts both cores to sleep.

My problem was, that `sleep_deep` takes some time during which the other core entered a critical_section interrupting  `sleep_deep` midway, choking on half powered down peripherals and never leaving the critical_section and thus `sleep_deep` never finished.

Maybe the sleep functions should require a `CriticalSection`?

bjoernQ, 2024-09-10:
Thanks - that's an interesting observation

blinry, 2024-10-29:
@dimpolo, I think I might be encountering a similar issue: #2423

Which microcontroller did you notice this on? Did you find a workaround?

dimpolo, 2024-10-29:
@blinry I believe this issue is not relevant to your case. In my case code running on the second core was messing with the first core going to sleep. In your example that is not possible.

To answer your question: chip: esp32s3, workaround: enter deep sleep from inside a critical_section

bugadani, 2026-09-08:
I don't know why we are tracking this as a bug - I don't see anything actionable here, except perhaps documenting things a bit better. Right now the general rule is that the other core should be stalled and interrupts disabled, which is done by the auto-sleep entry hook, but not by the manual sleep functions. Whether that is a good solution, is debatable, but this way you can prepare for sleep the way you need. Anyway, sleep is a system-wide state, it's not possible to sleep one core, but not the other.

closed 2026-09-08 by bugadani