esp-rs/esp-hal #2111: `esp-wifi`: Does not compile with feature `wifi-logs` any more.

frederictobiasc, 2024-09-06:
Hi

since v0.20.0, enabling `wifi-logs` feature on  `esp-wifi` produces the following error:

(Also happens on latest master d71434adfb7573c267d4d41a4304a2c556d59231)

```
    Compiling esp-wifi v0.9.1 (/home/istobic/develop/esp-hal/esp-wifi)
error[E0425]: cannot find function `vsnprintf` in this scope
  --> /home/istobic/develop/esp-hal/esp-wifi/src/compat/syslog.rs:10:13
   |
10 |             vsnprintf(&mut buf as *mut u8, 512, _format, _args);
   |             ^^^^^^^^^ not found in this scope
   |
help: consider importing this function
   |
1  + use esp_wifi_sys::include::vsnprintf;
   |

error[E0425]: cannot find function `str_from_c` in this scope
  --> /home/istobic/develop/esp-hal/esp-wifi/src/compat/syslog.rs:11:27
   |
11 |             let res_str = str_from_c(&buf as *const u8);
   |                           ^^^^^^^^^^ not found in this scope
   |
help: consider importing this function
   |
1  + use crate::compat::common::str_from_c;
   |

```

AnthonyGrondin, 2024-09-07:
Same behaviour is happening on esp32s3.

This bug was introduced in 5420ce09597cde5fa129d7cca62fb842158cf7ba

MabezDev, 2024-09-09:
Closed via #2117

closed 2024-09-09 by MabezDev