esp-rs/esp-hal #2116: Is it possible to add the internal temperature sensor support?

lu-zero, 2024-09-08:
not esp32 has support for it https://docs.espressif.com/projects/esp-idf/en/stable/esp32c3/api-reference/peripherals/temp_sensor.html

Twin issue for the esp-idf-hal crate https://github.com/esp-rs/esp-idf-hal/issues/333

MabezDev, 2024-09-17:
It's for sure possible, a PR would be most welcome :).

lu-zero, 2024-09-25:
I'd need lots of guidance first, I do not know enough of the esp-idf and esp-hal to commit to it yet ^^;

Hackswell, 2024-10-06:
Just some notes/thoughts.  For anyone wanting to implement.  Might make a good "beginner" task, since turning on is just a few bit flips, and you just read the lowest byte and follow a set algorithm.
   T(°C) = 0.4386 ∗ VALUE–27.88 ∗ offset–20.52

Basic low-level bit-flipping and definitions for the TSENSor are here:
   esp-pacs/esp32c3/src/apb_saradc/tsens_ctrl.rs
   esp-pacs/esp32c3/src/apb_saradc/tsens_ctrl2.rs

Hardware ports/memory are already defined in:
   esp-pacs/esp32c3/svd/esp32c3.base.svd

Would need to add the "tsens" peripheral here, I think
   esp-hal/esp-hal/src/system.rs
and possibly here?  Not sure:
   esp-hal/esp-hal/src/analog/adc/riscv.rs

Would it be better to add the higher level implementation here, since it's connected to the ADC peripheral?  But then it might get "hidden" and hard to find.
   esp-hal/esp-hal/src/analog/adc/esp32.rs

Or would it be better to implement this in
   esp-hal/esp-hal/src/analog/tsens/esp32.rs

closed 2025-01-07 by MabezDev