esp-rs/esp-hal #2142: Allow building different tagged docs

MabezDev, 2024-09-11:
Right now it's only possible to build docs from a single git revision. This is a problem when we release different crates at different times, case in point the latest esp-wifi release came a few days after the esp-hal one.

We're now in a situation where the docs are outdated for esp-wifi, but if we re-run the workflow, we'll now include breaking changes for esp-hal on the docs for the last release.

We probably need to extend the workflow to collate the different versions from different tags publish them. This means we need to start tagging other projects we want to document, at the moment this is just esp-wifi.

bjoernQ, 2024-09-26:
@MabezDev This can be closed? (Done in #2218 - or is there anything missing?)

MabezDev, 2024-09-26:
Oops! Thanks!

closed 2024-09-26 by MabezDev