esp-rs/esp-hal #2201: [ESP-WIFI] Index out of bounds panic when connecting to enterprise wifi.

m-dahl, 2024-09-20:
## Bug description

Hi. Thanks to the work in #2004 we have been able to switch our application from idf to esp-hal + embassy. Its working well but we hit an issue the other day. I have implemented simple roaming by scanning for better access points and disconnect the current one and selecting the one I want by setting a specific BSSID. It works fine, but after a while it panics (details below).

## To Reproduce

1. Connect and disconnect to wifi (enterprise) in a loop.
2. Panics on the 7th connection attempt.

e.g.
```rust
    controller.start().await.expect("wifi: start");
    controller.set_configuration(&client_conf).expect("wifi: set_configuration");
    loop {
        log::info!("CONNECT");
        controller.connect().await;
        Timer::after(Duration::from_millis(1000)).await;

        log::info!("DISCONNECT");
        controller.disconnect().await;
        Timer::after(Duration::from_millis(1000)).await;
    }
```

Result:
```
====================== PANIC ======================
panicked at /Users/user/.cargo/git/checkouts/esp-hal-42ec44e8c6943228/5de267a/esp-wifi/src/compat/common.rs:204:24:
index out of bounds: the len is 10 but the index is 10

Backtrace:

0x420233b6
0x420233b6 - esp_wifi::compat::common::create_recursive_mutex::{{closure}}
    at /Users/user/.cargo/git/checkouts/esp-hal-42ec44e8c6943228/5de267a/esp-wifi/src/compat/common.rs:204
```

It seems `MUTEX_IDX_CURRENT` is only ever incremented so eventually you go oob. Maybe related to warning `warn!("trying to remove a mutex via sem_delete");` which looks like a TODO in `common.rs`?

## Environment

- esp32c3 + enterprise wifi network
- esp-hal/wifi latest commit (5de267ab).

bjoernQ, 2024-09-20:
The issue is caused by wpa-supplicant which is really calling sem_delete for a mutex - I have a fix which I will PR soon.

However, while testing it turned out that on ESP32-C3 things still stop working after a few iterations while it works "forever" on other chips. We already discovered unrelated other issues so we need to update the wifi drivers 

I will start working on #2154 next, so we can get that fixed

closed 2024-09-20 by MabezDev

m-dahl, 2024-09-20:
Thanks for the quick update! I will follow the progress on #2154 then.

PR #2202 (2024-09-20): Create mutexes in heap memory
## Thank you for your contribution!

We appreciate the time and effort you've put into this pull request.
To help us review it efficiently, please ensure you've gone through the following checklist:

### Submission Checklist 📝
- [ ] I have updated existing examples or added new ones (if applicable).
- [ ] I have used `cargo xtask fmt-packages` command to ensure that all changed code is formatted correctly.
- [ ] My changes were added to the [`CHANGELOG.md`](https://github.com/esp-rs/esp-hal/blob/main/esp-hal/CHANGELOG.md) in the **_proper_** section.
- [ ] I have  [... cut]