esp-rs/esp-hal #2212: possibly bricking esp32-h2 thread border router

lattice0, 2024-09-23:
I tried creating a template project for esp32-h2 and after the first flash on this board on the esp32-h2 chip, it stops appearing as a usb device on linux. It keeps reconectting and appearing in the list of usb devices for 2 seconds only.

I first thought it was a problem with my board so I tried on another identical one and got the same problem

[image]

Did I do something wrong? I though it was a defect of the first board, but both behave like this.

I don't know if this is an official product, but at least the embedded esp32-h2 is. I also have standalone esp32-h2 which worked without any problems, so I wonder what is happening exactly

Link for the board https://pt.aliexpress.com/item/1005005232242277.html?gatewayAdapt=glo2bra

bugadani, 2024-09-23:
You may just be missing [this](https://github.com/esp-rs/esp-hal/blob/aacc001e5dbc0443abc2dd3cf14d7fafe195ffd9/esp-wifi/Cargo.toml#L121) feature

lattice0, 2024-09-23:
@bugadani so now that I already pushed the code without it, I cannot fix and it's bricked forever? Because it's rebooting before I can do anything with it.

Also, this does not happen with a standalone esp32-h2 devkit, it only happens with the thread border router with esp32-h2

Dominaezzz, 2024-09-23:
You can start it up in bootloader mode. Press and hold the boot button, then click the reset button, then let go of the boot button. 
This will let you flash a new program, regardless of what the previous one was doing

lattice0, 2024-09-23:
@bugadani wifi is not supported on the h2, but I did add the dependency anywasy
```

esp-wifi = { version = "0.8.0", features = [
    "esp32h2",
    "phy-enable-usb",
    "embedded-svc",
    "wifi-default",
] }
```

and got
```

 --- stderr
  thread 'main' panicked at /home/user/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-wifi-0.8.0/build.rs:51:9:

          WiFi is not supported on this target.

```

lattice0, 2024-09-23:
@bjoernQ indeed I was able to enter download mode. What can be done now?

lattice0, 2024-09-23:
Ok, I don't know why it stuck on download mode but I was able to make it flash again after pressing reset or something like that.

What is ˋphy-enable-usbˋ for? It looks like now I can see the UART output

bjoernQ, 2024-09-23:
ˋphy-enable-usbˋ would allow USB to continue to work when the radio peripherals are enabled - but if you didn't use Bluetooth or IEEE802154 that wasn't the problem. Good to hear you are able to flash new firmwares again

MabezDev, 2024-09-25:
Seems like this is resolved - closing.

closed 2024-09-25 by MabezDev