esp-rs/esp-hal #2217: esp-wifi Exception StoreProhibited

Hero9909, 2024-09-24:
## Bug description

i tried to use the latest state and upon initialization the showen error occured. 

the changes made in the last 3 weeks are too big for me to describe futher whats probably the issues. but on the stable version the initialization works fine to me, but i cannot say if it worked futher as i was experimenting.

## To Reproduce

1. generate a nostd  project for the esp32
2. add embassy for  async 
3. add esp-wifi with ble, async and esp32 features

## Expected behavior

normal initialization as it works in the stable release

## Environment

- Target device: ESP32
- Crate name and version: esp-wifi:26fc8871db28d12ac5269e8297baf7d43a8c59d4 (latest at time of writing)

### Trace
```

WARN - Unable to allocate 144 bytes

Exception occurred 'StoreProhibited'
Context
PC=0x400e088b       PS=0x00060335
0x400e088b - esp_wifi::preempt::allocate_main_task::{{closure}}
    at /home/user/.cargo/git/checkouts/esp-hal-42ec44e8c6943228/26fc887/esp-wifi/src/preempt/mod.rs:26
A0=0x800dfc43       A1=0x3ffdb320       A2=0x400e9e78       A3=0x3ffc33d4       A4=0x00000001
0x400e9e78 - esp_hal::system::PeripheralClockControl::enable
    at /home/user/.cargo/git/checkouts/esp-hal-42ec44e8c6943228/26fc887/esp-hal/src/system.rs:129
0x3ffc33d4 - esp_wifi::preempt::CTX_NOW.1
    at ??:??
A5=0x0000ffff       A6=0x00060320       A7=0x3ffc33d0       A8=0x800e088b       A9=0x3ffdb2a0
0x3ffc33d0 - esp_wifi::preempt::CTX_NOW.0
    at ??:??
A10=0x00000000      A11=0x00000090      A12=0x3ffdb720      A13=0x00000000      A14=0x3ffc0000
A15=0x00000090
SAR=00000011
EXCCAUSE=0x0000001d EXCVADDR=0x00000088
LBEG=0x4000c46c     LEND=0x4000c477     LCOUNT=0x00000000
THREADPTR=0x00000000
SCOMPARE1=0x00000100
BR=0x00000000
ACCLO=0x00000000    ACCHI=0x00000000
M0=0x00000000       M1=0x00000000       M2=0x00000000       M3=0x00000000
F64R_LO=0x00000090  F64R_HI=0x3f401f03  F64S=0x00000004
FCR=0x3f404d94      FSR=0x00000005
F0=0x3ffdb2c8       F1=0x00000004       F2=0x00000000       F3=0x2a707c2b       F4=0x3ffdb2a0
F5=0x400eb1a8       F6=0x3ffdb2e8       F7=0x400ee048       F8=0x3ffdb2ec       F9=0x400eb1bc
0x400eb1a8 - <&T as core::fmt::Display>::fmt
    at /home/user/.rustup/toolchains/esp/lib/rustlib/src/rust/library/core/src/fmt/mod.rs:2368
0x400ee048 - log::Level::as_str
    at /home/user/.cargo/registry/src/index.crates.io-6f17d22bba15001f/log-0.4.22/src/lib.rs:566
0x400eb1bc - <core::fmt::Arguments as core::fmt::Display>::fmt
    at /home/user/.rustup/toolchains/esp/lib/rustlib/src/rust/library/core/src/fmt/mod.rs:478
F10=0x3ffdb2a8      F11=0x400eb1a8      F12=0x00000003      F13=0x3ffdb72c      F14=0x3ff0001c
0x400eb1a8 - <&T as core::fmt::Display>::fmt
    at /home/user/.rustup/toolchains/esp/lib/rustlib/src/rust/library/core/src/fmt/mod.rs:2368
F15=0x3ff00020
0x400e5896
main
    at ??:??
0x400ef801
Reset
    at ??:??
0x400ea296
ESP32Reset
    at ??:??
```

bjoernQ, 2024-09-24:
That is very, very early during initialization (actually the very first memory allocation) - are you setting up a heap to be used by esp-wifi like shown here: 

https://github.com/esp-rs/esp-hal/blob/26fc8871db28d12ac5269e8297baf7d43a8c59d4/examples/src/bin/wifi_dhcp.rs#L52

Before esp-wifi was setting up and managing its own heap but now it a heap shared and initialized by the application

bjoernQ, 2024-09-24:
Great to hear it's working now.

The size depends on if your code is also doing dynamic allocations, which chip (e.g. C6 needs slightly more memory) and which features are used (WiFi vs BLE vs both). As a rule of thumb if you see a lot of failed allocations try with a bit more memory

closed 2024-09-24 by MabezDev