esp-rs/esp-hal #2222: Floating point operation crashes with "Exception occurred 'Cp0Disabled'" while connected to a Wifi

Cutkh, 2024-09-24:
## Bug description

The code below crashes on an ESP32 (ESP-WROOM-32) with the exception below.
The issue seems to be the f32/float operation in the "read_sensor_data()" function, while being connected to a wifi.
 If I remove the wifi part or set invalid wifi connection data, it performs the floating point operations normally.

```

Exception occurred 'Cp0Disabled (Access to the floating point coprocessor is not allowed. You may want to enable the `float-save-restore` feature of the `xtensa-lx-rt` crate.)'
Context
PC=0x400f00b7       PS=0x00060930
0x400f00b7 - core::f32::<impl f32>::abs_private
    at /home/foobar/.rustup/toolchains/esp/lib/rustlib/src/rust/library/core/src/num/f32.rs:533
0x00060930 - PS_WOE
    at ??:??
A0=0x800ee5d5       A1=0x3ffdab00       A2=0x3ffdb638       A3=0x3ffdab30       A4=0x00000000
0x800ee5d5 - _rtc_slow_bss_start
    at ??:??
0x3ffdab00 - __stack_chk_guard
    at ??:??
0x3ffdb638 - __stack_chk_guard
    at ??:??
0x3ffdab30 - __stack_chk_guard
    at ??:??
0x00000000 - <esp32::RMT as core::fmt::Debug>::fmt
    at /home/foobar/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp32-0.33.0/src/lib.rs:1533
A5=0x0006ffff       A6=0x00000000       A7=0x00000001       A8=0x00000001       A9=0x4442accd
0x0006ffff - PS_WOE
    at ??:??
0x00000000 - <esp32::RMT as core::fmt::Debug>::fmt
    at /home/foobar/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp32-0.33.0/src/lib.rs:1533
0x00000001 - <esp32::RMT as core::fmt::Debug>::fmt
    at /home/foobar/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp32-0.33.0/src/lib.rs:1533
0x00000001 - <esp32::RMT as core::fmt::Debug>::fmt
    at /home/foobar/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp32-0.33.0/src/lib.rs:1533
0x4442accd - wDev_GetNoiseFloor
    at ??:??
A10=0x7fffffff      A11=0x00000000      A12=0x00000000      A13=0x3ffdbd50      A14=0x00000010
0x7fffffff - _rtc_slow_bss_start
    at ??:??
0x00000000 - <esp32::RMT as core::fmt::Debug>::fmt
    at /home/foobar/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp32-0.33.0/src/lib.rs:1533
0x00000000 - <esp32::RMT as core::fmt::Debug>::fmt
    at /home/foobar/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp32-0.33.0/src/lib.rs:1533
0x3ffdbd50 - __stack_chk_guard
    at ??:??
0x00000010 - <esp32::RMT as core::fmt::Debug>::fmt
    at /home/foobar/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp32-0.33.0/src/lib.rs:1534
A15=0x00000001
0x00000001 - <esp32::RMT as core::fmt::Debug>::fmt
    at /home/foobar/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp32-0.33.0/src/lib.rs:1533
SAR=00000004
EXCCAUSE=0x00000020 EXCVADDR=0x00000000
0x00000020 - <esp32::RMT as core::fmt::Debug>::fmt
    at /home/foobar/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp32-0.33.0/src/lib.rs:1535
0x00000000 - <esp32::RMT as core::fmt::Debug>::fmt
    at /home/foobar/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp32-0.33.0/src/lib.rs:1533
LBEG=0x4000c28c     LEND=0x4000c296     LCOUNT=0x00000000
0x4000c28c - memcmp
    at ??:??
0x4000c296 - memcmp
    at ??:??
0x00000000 - <esp32::RMT as core::fmt::Debug>::fmt
    at /home/foobar/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp32-0.33.0/src/lib.rs:1533
THREADPTR=0x00000000
0x00000000 - <esp32::RMT as core::fmt::Debug>::fmt
    at /home/foobar/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp32-0.33.0/src/lib.rs:1533
SCOMPARE1=0x00000003
0x00000003 - <esp32::RMT as core::fmt::Debug>::fmt
    at /home/foobar/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp32-0.33.0/src/lib.rs:1533
[... 295 lines cut ...]
xtensa-lx = {version="0.9.0"}
xtensa-lx-rt = {version="0.17.1",feature = ["esp32", "float-save-restore"]}

critical-section    = "1.1.2"

rust-mqtt = {version="0.3.0", default-features = false}
libm = "0.2.8"
postcard = "1.0.10"
serde-json-core = "0.6.0"

[profile.dev]
# Rust debug is too slow.
# For debug builds always builds with some optimization
opt-level = "s"

[profile.release]
codegen-units = 1 # LLVM can perform better optimizations using a single thread
debug = 2
debug-assertions = false
incremental = false
lto = 'fat'
opt-level = 's'
overflow-checks = false

```

Dominaezzz, 2024-09-24:
See https://github.com/esp-rs/esp-hal/issues/2050#issuecomment-2323010837

Cutkh, 2024-09-24:
The feature 'float-save-restore' flag is set. as you can see in the Cargo.toml above.

Dominaezzz, 2024-09-24:
Ah I see. I'm out of ideas then :slightly_smiling_face:

bjoernQ, 2024-09-24:
There is a typo here `xtensa-lx-rt = {version="0.17.1",features = ["esp32", "float-save-restore"]}`

In your code above it's "feature" but needs to be "features"

closed 2024-09-24 by MabezDev

Cutkh, 2024-09-25:
Woa, I've wasted three hours on this and double-checked the Cargo.toml three times - but missed this completely...
Now it's working like a charm :-D
Thank you!