esp-rs/esp-hal #2246: DHCP takes a long time to get an IP address

bjoernQ, 2024-09-27:
Sometimes it takes quite long to get an IP address

Apparently, it takes a few Discover messages until my AP sends a Request - with an ESP-IDF example that is not the case:

[image]

Here it took five discover messages until I got a request. The next discover+request was from esp-idf

The strangest thing is that the transaction id of the request doesn't match any discover message (but the IP was accepted). For ESP-IDF everything looked normal

I even changes smoltcp to send the exact same request but nothing changed.

It might be something special about my AP (so would be good if others could verify that)

bjoernQ, 2024-09-27:
Seems like smoltcp will always generate a new transaction ID when it sends a message - which is ok and explains the different transaction id in the request

bugadani, 2024-09-27:
Is this async? If so, we had a similar problem last year where the embassy-net task wasn't waken.

bjoernQ, 2024-09-27:
That's using the blocking API to keep the testing as simple as possible

AnthonyGrondin, 2024-09-27:
From personal experience, I have noticed a different behavior for getting an IP address between two esp32s3.

One is an esp32-s3-mini-1, on a breadboard, and it takes a long time to get connected to wifi and get an IP.
The other is an esp32-s3-wroom-1, mounted on a PCB and it always connects faster and get an IP fast. I assume this is due to an antenna behavior, as the PCB acts as an antenna, so I never looked at the Wireshark trace.

I flashed the `wifi_dhcp` example on both 

Here is the capture for the esp32-s3-mini-1 on a breadboard: 
[image]

Here is the capture for the esp32-s3-wroom-1 installed on a PCB:
[image]

bjoernQ, 2024-09-27:
I did some more testing - this time using my laptop as an AP

[image]

Looks exactly the same for blocking and async. Looks very good but then it waits five seconds and sends the request again.

PR #2251 (2024-10-04): esp-wifi: Align default settings
## Thank you for your contribution!

We appreciate the time and effort you've put into this pull request.
To help us review it efficiently, please ensure you've gone through the following checklist:

### Submission Checklist 📝
- [ ] I have updated existing examples or added new ones (if applicable).
- [x] I have used `cargo xtask fmt-packages` command to ensure that all changed code is formatted correctly.
- [ ] My changes were added to the [`CHANGELOG.md`](https://github.com/esp-rs/esp-hal/blob/main/esp-hal/CHANGELOG.md) in the **_proper_** section.
- [ ] I ha [... cut]
closed 2024-10-04 by bugadani