esp-rs/esp-hal #2273: Peripheral signal API does not force signals to be routed through the GPIO matrix

bugadani, 2024-10-03:
This could result in some funny behaviour if the user expect to loop back a GPIO into a peripheral, but the signal is not going through the GPIO matrix.

Dominaezzz, 2024-10-03:
If I remember correctly you don't need the GPIO matrix for loopback/interconnect.

It should be fine for the output to go via io_mux and input to go via matrix, so long as the GPIO pad has both input and output enabled. (Which you have to do in the right order based on the current GPIO code)

bugadani, 2024-10-03:
You kinda do, because you can't select two alternate functions for the same pin. I spent more time on this today than I'd like to admit.

Dominaezzz, 2024-10-03:
Oh, I'd always thought of it as selecting an alternative function for the pin and an alternative function for the input signal, TIL

closed 2024-11-06 by MabezDev