esp-rs/esp-hal #2274: esp-wifi: Investigate time to get IP address from DHCP

JurajSadel, 2024-10-04:
Related to https://github.com/esp-rs/esp-hal/issues/2246. There seems to be another problem which is not handled (and probably should be handled separately) in https://github.com/esp-rs/esp-hal/pull/2251. This issue aims for tracking and investigating it.

bjoernQ, 2024-10-23:
One user is noticing problems when using `phy-enable-usb`: https://github.com/esp-rs/esp-hal/discussions/2345#discussioncomment-11022245

Same user observes more problems when using a systimer alarm as async timer compared to TIMG (https://github.com/esp-rs/esp-hal/discussions/2391)

bjoernQ, 2024-10-30:
Tested connecting to another ESP32(-C3) running a variant of https://github.com/bjoernQ/esp32c3-dhcp-experiment (just removed everything but DHCP) and ESP32-C6 was able to always get an IP in 50ms - 400ms (async slightly better)

bjoernQ, 2024-10-30:
Tested ESP32-C6 with hot-spot running on Windows 11: 200 - 400ms

Using a Fritz!Box 5590: 20 - 30 ms

MabezDev, 2024-10-30:
I'm wondering if in a congested network esp-wifi can somehow miss/drop packets - it seems on a quiet network (mine & @bjoernQ) it seems to work quite well but in the office not so much. I'm just guessing at this point but that seems to be the only real difference :thinking:

bjoernQ, 2024-10-30:
(when I was in the office, I also tested it briefly - no problems at all)

bjoernQ, 2024-10-30:
Oh ... all the testing above was on current main ( 8a23dbe1b6e10459e4d7456f2da96b4c406e7740 ) - on the `more-dynamic-allocations` branch the numbers a worse

bjoernQ, 2024-10-30:
False alarm caused by unintendedly enabling ps-max-modem in that branch (but proves that using modem sleep makes getting the IP address harder)

jessebraham, 2024-11-07:
What's the status here? Some people seem to be seeing issues, but repeatedly when trying to test things seem to be working on our end. Is this just a signal strength or network congestion issue? Is there anything actionable here, or should we just close this for now?

bjoernQ, 2024-11-07:
I don't see anything I can do at least (or anything we can do by changing code at all) - I spent almost a day trying out multiple scenarios - even bad reception and multiple APs using the same SSID etc.

DHCP server sends a single broadcast - if we don't receive it, we wait for the timeout (IIRC 10 seconds) and try again until we succeed.

Timing probably makes a huge difference by randomly working better or worse for different setups I guess

So - if no one comes up with a great idea what to do here I'd say we should close it

closed 2024-11-07 by MabezDev