esp-rs/esp-hal #2277: Async RMT seems to produce incorrect output

JurajSadel, 2024-10-04:
Blocking and async example don't produce the same output signal. Tested with S3 and C6, most likely all chips are affected.

JurajSadel, 2024-10-04:
```rust
//! Demonstrates decoding pulse sequences with RMT
//!
//! The following wiring is assumed:
//! - Connect GPIO4 and GPIO5

//% CHIPS: esp32 esp32c3 esp32c6 esp32h2 esp32s2 esp32s3
//% FEATURES: embassy embassy-generic-timers

#![no_std]
#![no_main]

use embassy_executor::Spawner;
use embassy_time::{Duration, Timer};
use esp_backtrace as _;
use esp_hal::{
    delay::Delay,
    gpio::{Io, Level, Output},
    prelude::*,
    rmt::{asynch::RxChannelAsync, PulseCode, Rmt, RxChannelConfig, RxChannelCreatorAsync},
    timer::timg::TimerGroup,
};
use esp_println::{print, println};

const WIDTH: usize = 80;

#[cfg(is_not_release)]
compile_error!("Run this example in release mode");

#[embassy_executor::task]
async fn signal_task(mut out: Output<'static>, delay: Delay) {
    
    
    loop {
        for i in 0u32..5u32 {
            out.set_high();
            delay.delay_micros(i * 10 + 20);
            out.set_low();
            delay.delay_micros(i * 20 + 20);
        }
    }
}

#[esp_hal_embassy::main]
async fn main(spawner: Spawner) {
    println!("Init!");
    let peripherals = esp_hal::init(esp_hal::Config::default());

    let timg0 = TimerGroup::new(peripherals.TIMG0);
    esp_hal_embassy::init(timg0.timer0);

    let io = Io::new(peripherals.GPIO, peripherals.IO_MUX);
    // let mut out = Output::new(io.pins.gpio5, Level::Low);

    cfg_if::cfg_if! {
        if #[cfg(feature = "esp32h2")] {
            let freq = 32.MHz();
        } else {
            let freq = 80.MHz();
        }
    };

    let delay = Delay::new();

    let rmt = Rmt::new_async(peripherals.RMT, freq).unwrap();
    let rx_config = RxChannelConfig {
        clk_divider: 1,
        idle_threshold: 10000,
        ..RxChannelConfig::default()
    };

    cfg_if::cfg_if! {
        if #[cfg(any(feature = "esp32", feature = "esp32s2"))] {
            let mut channel = rmt.channel0.configure(io.pins.gpio4, rx_config).unwrap();
        } else if #[cfg(feature = "esp32s3")] {
            let mut channel = rmt.channel7.configure(io.pins.gpio4, rx_config).unwrap();
        } else {
            let mut channel = rmt.channel2.configure(io.pins.gpio4, rx_config).unwrap();
        }
    }

    spawner
        .spawn(signal_task(Output::new(io.pins.gpio5, Level::Low), delay))
        .unwrap();

    let mut data = [PulseCode {
        level1: true,
        length1: 1,
        level2: false,
        length2: 1,
    }; 48];

    loop {
        // for x in data.iter_mut() {
        //     x.length1 = 0;
        //     x.length2 = 0;
        // }

         // simulate input
        //  for i in 0u32..5u32 {
        //     out.set_high();
        //     delay.delay_micros(i * 10 + 20);
        //     out.set_low();
        //     delay.delay_micros(i * 20 + 20);
        // }
        
        println!("receive");
        channel.receive(&mut data).await.unwrap();
        let mut total = 0usize;
        println!("data_leng :{:?}", data.len()); // NOT PRINTED

        for entry in &data[..data.len()] {
            println!("test");

            if entry.length1 == 0 {
                break;
            }
            total += entry.length1 as usize;

            if entry.length2 == 0 {
                break;
            }
            total += entry.length2 as usize;
        }

        for entry in &data[..data.len()] {
            if entry.length1 == 0 {
                break;
            }

            let count = WIDTH / (total / entry.length1 as usize);
            let c = if entry.level1 { '-' } else { '_' };
            for _ in 0..count + 1 {
                print!("{}", c);
            }

            if entry.length2 == 0 {
                break;
            }

            let count = WIDTH / (total / entry.length2 as usize);
            let c = if entry.level2 { '-' } else { '_' };
            for _ in 0..count + 1 {
                print!("{}", c);
            }
        }

        println!();
    }
}
```
prints
```rust
Init!
receive
```
seems it is stuck in `channel.receive(&mut data).await.unwrap();`

bjoernQ, 2024-10-04:
The signal_task contains no await-point so the main task is starving

JurajSadel, 2024-10-07:
Everything seems to be working fine now, sorry for the noise.

closed 2024-10-07 by JurajSadel