esp-rs/esp-hal #2285: Initial P4 support tracking issue

MabezDev, 2024-10-07:
Whilst we don't intend to do a full chip bring-up until the mass production version is released, we would like to get a minimal chip bring-up to aid future developments.

- [ ] PAC support
- [ ] Linker and required `soc` module features
- [ ] Tooling(?)
- [ ] Uart driver

jessebraham, 2024-10-09:
I again have a very basic application running on device, calling only `esp_println::println!()`.

There is a caveat to this, in that the application works on the `v1.0` devkit but neither the `v1.1` not `v1.2` version; I have some ideas with regards to this, but will require more experimentation. At least for the time being, I'm able to move forward using the old devkit.

Unfortunately I think getting a UART example running will take a bit more time than expected, as there is quite tight coupling between the interrupt support and various drivers. Given that the ESP32-P4 uses yet another interrupt controller (CLIC) this will take some time to sort out. Hoping to have a hacky PoC by end of week or early next week, though.

I will continue to update this issue as I make progress.

bjoernQ, 2024-10-09:
I remember we had interrupts and gpio working in January or February already

jessebraham, 2024-10-10:
> I remember we had interrupts and gpio working in January or February already

Yes but there have been a number of changes to the HAL since January, and it's not as simple as copy-pasting the previous code in unfortunately. I do not think this will take a ton of time to be clear, but it will take some time to understand the changes which have been made and adapt the previous implementation to match.

playfulFence, 2024-10-30:
Working on this right now. I'm not sure how far I'll get, but at least I want to adapt all the changes to the current state of a driver.

jessebraham, 2024-11-08:
@playfulFence Is the P4 HIL runner in a functional state? Can we tick off that task?

bugadani, 2024-11-08:
Define functional, we have a raspberry that boots. We don't have anything on it, let alone flashing support in probe-rs or GHA workflows in this repo. What's the definition of done of that checkbox?

jessebraham, 2024-11-08:
> Define functional, we have a raspberry that boots. We don't have anything on it, let alone flashing support in probe-rs or GHA workflows in this repo. What's the definition of done of that checkbox?

Been awhile since I opened this, but given that there is a separate task for tooling (which links to the `probe-rs` issue), I think just having the runner set up physically and having the GitHub runner configured is adequate.

MabezDev, 2024-11-08:
Dropping this back to the current milestone, there's a chance we can complete it in time, but if not we'll reassess what to do in the next planning meeting.

MabezDev, 2024-11-14:
@playfulFence has been working hard to get interrupt support working. It's _nearly_ there, and works correctly _without_ vectoring the CPU interrupts (this is the RISCV interrupt controller vectoring, different from vectoring the peripheral interrupts). The P4 is quite different to the other esp's, and requires a number of changes in esp-riscv-rt (based on esp-idf, we didn't actually get this working just yet).

I think we'll want to revisit this support once https://github.com/esp-rs/esp-hal/issues/2390 has landed, as upstream riscv-rt has better support for this and we should be directing our efforts there anyway.

We'll leave the PR open https://github.com/esp-rs/esp-hal/pull/2466 as draft, and we can occasionally rebase if anyone would like to try out running some pure Rust code on a early P4 devkit.

rmsyn, 2025-08-04:
> Sounding off that I'm interested in helping, and I'll start pointing people here when they ask.

Same, happy to help out with testing and dev.

SergioGasquez, 2025-08-28:
Hi @jamesmunns, @rmsyn, here is our current chip support plan: #3962 with updated p4 plans, if you have any questions, just let me know

rmsyn, 2025-08-28:
> The analogy would be two somewhat different PAC crates for us.

So, any HAL work would need to potentially use conditional compilation to handle register naming differences in the PACs?

vliberio, 2026-01-28:
Yeah I was (and maybe am) still a little confused myself on P4's mass production & distribution status and contacted Espressif (sales@espressif.com) last month and got a couple emails back. I'll put the relevant responses here in case it helps anyone.

My Questions:
```
1. Are SoCs such as the ESP32-P4NRW32 the full, supported, production-ready ESP32-P4 product? The 'Marketing Status' in the Product Selector indicates these are 'Mass Production', but the datasheet from the same Product Selector page indicates it is 'Pre-release v0.6' and has 'PRELIMINARY' watermarked over each page? Are these just a sample / pre-release variant of the ESP32-P4?

2. If these are these are fully supported production ESP32-P4 variants, is there an official vendor / distributor for the ESP32-P4 that we could obtain a small batch (say 50-100?) for prototypes? The only listing we've so far seen was on LCSC (https://www.lcsc.com/product-detail/C22387510.html). Is LCSC an authorized distributor or is there one that offers the ESP32-P4 in the US?

3. Will the ESP32-P4 receive limited or full support for the official Rust SDK in the foreseeable future like with the other boards? (https://developer.espressif.com/blog/2025/10/esp-hal-1/)
```

Response #1 Snippet:
```
For question 2, LCSC (https://www.lcsc.com/product-detail/C22387510.html) is not one of  our authorized distributors. You can obtain ESP32-P4 samples through Espressif’s official store on AliExpress via the following link: https://www.aliexpress.com/item/1005008985298657.html 

Regarding questions 1 and 3, we will provide you with further responses as soon as possible. Thank you.
```

Response #2 Snippet:
```
1.	ESP32-P4NRW32 is a fully supported, production-ready ESP32-P4 SoC, and the “Mass Production” status in the Product Selector is correct.
Sorry, we still have some internal details to finalize. Once these are completed, we will release the v1.0 version of the datasheet as soon as possible.
2.	Our business colleagues have already provided a response in the previous email. Please refer to it.
3.	I have already consulted the relevant colleagues, and if there is any reply, I will immediately get back to you via email.
```

Personally, I was waiting for the datasheet to get finalized before proceeding further.

Avamander, 2026-09-17:
@MabezDev What about the relatively widespread ESP32-P4 rev1/rev2 chips being supported?

I'm pretty sure the current HAL doesn't account for the differences and it's not a feature one could toggle. Things such as the non-standard CLIC, different ROM, clock speed, etc.

bugadani, 2026-09-17:
As stated in https://github.com/esp-rs/esp-hal/issues/3962, chips before v3 will not be supported by us.