esp-rs/esp-hal #2289: `embassy_interrupt_spi_dma`: C2, C3, C6, H2 - need to use a different DMA-enabled peripheral

tom-borcin, 2024-10-07:


bjoernQ, 2024-11-06:
I'm not quite sure what this issue is asking for. The mentioned targets only contain one GP-SPI

While it's technically possible to use SPI1 we don't support that and I'm not sure if we ever will or should (I'd personally say we shouldn't)

bugadani, 2024-11-06:
The idea is to use I2S on these devices (or all of them).

bjoernQ, 2024-11-06:
> The idea is to use I2S on these devices (or all of them).

Ok nice - C2 doesn't include I2S IIRC but at least this clarifies the issue a bit

bugadani, 2024-11-06:
To be honest this is a bit of a non-issue anyway. With the Xtensas we have a test for both PDMA and GDMA devices and the original issue wasn't exactly device-specific.

bjoernQ, 2024-11-08:
Hint: the PR which introduced the test is https://github.com/esp-rs/esp-hal/pull/2065

PR #2490 (2024-11-08): Test more embassy interrupt spi dma
## Thank you for your contribution!

We appreciate the time and effort you've put into this pull request.
To help us review it efficiently, please ensure you've gone through the following checklist:

### Submission Checklist 📝
- [ ] I have updated existing examples or added new ones (if applicable).
- [x] I have used `cargo xtask fmt-packages` command to ensure that all changed code is formatted correctly.
- [ ] My changes were added to the [`CHANGELOG.md`](https://github.com/esp-rs/esp-hal/blob/main/esp-hal/CHANGELOG.md) in the **_proper_** section.
- [ ] I [... cut]
closed 2024-11-08 by JurajSadel