esp-rs/esp-hal #2299: move `WifiStack` to seperate crate or use a different stack

MabezDev, 2024-10-08:
Feels a bit odd that it's included in esp-wifi itself, given that our/other BLE stack is separate.

I think we should either extract it, or consider another blocking stack to use for those examples.

bjoernQ, 2024-10-08:
👍 definitely yes - in the very early days we used `smoltcp-nal` but it was notoriously behind the current smoltcp version and was/is (?) a bit lacking in terms of functionality (e.g. server sockets) .... probably worth to have another look or just move the code to somewhere for now

Frostie314159, 2024-10-08:
In doing this, I'd suggest moving the phy init to a separate crate, since it's shared between all rf peripherals.

bjoernQ, 2024-11-06:
> In doing this, I'd suggest moving the phy init to a separate crate, since it's shared between all rf peripherals.

Mind opening a separate issue for this?

PR #2488 (2024-11-12): Remove the integrated blocking networking stack
## Thank you for your contribution!

We appreciate the time and effort you've put into this pull request.
To help us review it efficiently, please ensure you've gone through the following checklist:

### Submission Checklist 📝
- [x] I have updated existing examples or added new ones (if applicable).
- [x] I have used `cargo xtask fmt-packages` command to ensure that all changed code is formatted correctly.
- [x] My changes were added to the [`CHANGELOG.md`](https://github.com/esp-rs/esp-hal/blob/main/esp-hal/CHANGELOG.md) in the **_proper_** sect [... cut]
closed 2024-11-12 by jessebraham