esp-rs/esp-hal #2313: `dma_mem2mem`: S2

SergioGasquez, 2024-10-09:


Dominaezzz, 2024-10-10:
Looks like the PAC is missing the registers for Copy DMA. It's also missing the registers for AES/SHA/ADC DMA.

It's also worth noting that AES/SHA (Crypto DMA), SPI2 and I2S can also be used for memcpy operations.
I was planning to only add support for the dedicated copy DMA channel but it seems it doesn't support accessing external ram, but the non-dedicated ones do support external ram access :smile:, so supporting using those channels as well may be handier than I anticipated. (The only complication is it doesn't play nice with DMA channel erasure, maybe this driver can be special, since it kinda is :slightly_smiling_face:).

bugadani, 2024-10-10:
> The only complication is it doesn't play nice with DMA channel erasure, maybe this driver can be special, since it kinda is

Not everything needs to be erased, and I think we can even transform channels into diferent types for m2m if we want to. Don't worry about this part.

There is some investigation going on internally whether ESP32 peripheral DMAs support m2m as they do on the S2. Probably not, but it's not obvious at this point I think.

Dominaezzz, 2024-10-11:
Unassigned myself, not sure I'm too keen on learning how to update the PACs right now.

bugadani, 2024-11-20:
FYI https://github.com/esp-rs/esp-pacs/pull/302

Dominaezzz, 2025-02-02:
https://github.com/esp-rs/esp-pacs/issues/312

Dominaezzz, 2025-02-06:
https://github.com/esp-rs/esp-pacs/issues/318

PR #3352 (2025-04-09): Add DMA memcpy support to the S2
## Thank you for your contribution!

We appreciate the time and effort you've put into this pull request.
To help us review it efficiently, please ensure you've gone through the following checklist:

### Submission Checklist 📝
- [x] I have updated existing examples or added new ones (if applicable).
- [x] I have used `cargo xtask fmt-packages` command to ensure that all changed code is formatted correctly.
- [x] My changes were added to the [`CHANGELOG.md`](https://github.com/esp-rs/esp-hal/blob/main/esp-hal/CHANGELOG.md) in the **_proper_** section.
- [x] I ha [... cut]