esp-rs/esp-hal #2347: Can't manually create `esp_hal::Config` struct

Sycrosity, 2024-10-13:
## Bug description

When using `esp_hal::init`, like when wanting to change the clock speed of the chipm you cannot pass in a manually constructerd `esp_hal::Config` struct due to the `#[non_exhaustive]` tag.

## To Reproduce

1. On any esp_hal project/example, replace `let peripherals = esp_hal::init(esp_hal::Config::default());` with 
```rust
    let config = esp_hal::Config {
        cpu_clock: CpuClock::max(),
        watchdog: WatchdogConfig::default()
    };

    let peripherals = esp_hal::init(config);
```
2. Run the project and see the error:
```console

error[E0639]: cannot create non-exhaustive struct using struct expression
  --> src/main.rs:49:18
   |
49 |       let config = esp_hal::Config {
   |  __________________^
50 | |         cpu_clock: CpuClock::max(),
51 | |         watchdog: WatchdogConfig::default()
52 | |     };
   | |_____^
```

## Expected behavior

Either the struct is not non-exhaustive, or the struct has constructor methods availible, such as 
```rust

let peripherals = esp_hal::init(esp_hal::Config::new(CpuClock::max(),WatchdogConfig::default()));
```
or
```rust
let peripherals = esp_hal::init(esp_hal::Config::new().with_cpuclock(CpuClock::max()));

```

## Environment

- Target device: esp32c3
- Crate name and version: esp-hal v0.21.0

bugadani, 2024-10-13:
This is a somewhat unfortunate side of something that was done intentionally - to prevent adding stuff to Config becoming breaking changes. Duplicate of https://github.com/esp-rs/esp-hal/issues/2086. You can do something like this:

```rust
let peripherals = esp_hal::init({
    let mut config = esp_hal::Config::default();
    config.cpu_clock = CpuClock::max();
    config
});
```

closed 2024-10-13 by bugadani

Sycrosity, 2024-10-13:
Ah thank you yes that works fine for now. Ideally a constructor would be better than this method, would that be worth opening an issue for?

bugadani, 2024-10-13:
We don't intend to add code here just yet, because we're unsure how big the struct will end up in the long run. If we add a constructor for cpu clock and wdt config, then we remove it one release down the line or add a new argument to it (which is, being unnamed, a very unfriendly change), we've defeated the purpose of making the struct `#[non_exhaustive]`.