esp-rs/esp-hal #2358: Unable to connect to WiFi access point, or subscribe to advertised BLE

jsprog, 2024-10-16:
## Bug description

I had success with `wifi_dhcp.rs`, `wifi_embassy_dhcp`, `wifi_bench`, and `wifi_embassy_bench` examples on `esp32-c3` [https://www.wemos.cc/en/latest/c3/c3_pico.html](https://www.wemos.cc/en/latest/c3/c3_pico.html), not with other examples for WiFi and/or BLE.
- joining the `esp-wifi` take a while (connecting) until rejection from the client.
- pairing to BLE advertised esp32-c3 not making progress until rejection by the other device. Pressing the boot-bottom won't report anything about subscription state.

## To Reproduce

Currently, the main branch is the release version 0.21.1. After setting the right environment in .cargo/config.toml, I run some examples:

- `cargo xtask run-example esp-hal esp32c3 wifi_access_point`
- `cargo xtask run-example esp-hal esp32c3 wifi_embassy_access_point`
- `cargo xtask run-example esp-hal esp32c3 wifi_ble`
- `cargo xtask run-example esp-hal esp32c3 wifi_embassy_ble`

## Expected behavior

- The wireless stack should make progress when connecting to `esp-wifi` or pairing with the advertised BLE `esp32c3`.
- I already suspected my hardware until I run a micropython access-point example with success
  - MicroPython firmware: [https://github.com/wemos/micropython/releases/download/v1.23.0-220-g6927639898/firmware-LOLIN_C3_PICO-v1.23.0-220-g692763989.bin](https://github.com/wemos/micropython/releases/download/v1.23.0-220-g6927639898/firmware-LOLIN_C3_PICO-v1.23.0-220-g692763989.bin)
  - MicroPython example from this link [https://randomnerdtutorials.com/micropython-esp32-esp8266-access-point-ap/](https://randomnerdtutorials.com/micropython-esp32-esp8266-access-point-ap/)

## Environment
- `espflash 3.1.1`
- `ESP32-C3FH4` [https://www.wemos.cc/en/latest/c3/c3_pico.html](https://www.wemos.cc/en/latest/c3/c3_pico.html)
```
$: espflash board-info
[2024-10-16T12:37:28Z INFO ] Serial port: '/dev/ttyACM0'
[2024-10-16T12:37:28Z INFO ] Connecting...
[2024-10-16T12:37:28Z INFO ] Using flash stub
Chip type:         esp32c3 (revision v0.3)
Crystal frequency: 40 MHz
Flash size:        4MB
Features:          WiFi, BLE
MAC address:       68:67:25:b7:8b:b0
```
- esp-hal 0.21.1

bjoernQ, 2024-10-16:
Please note the access point examples DON'T include a DHCP server - did you make sure to use a static IP? Otherwise, the station will most probably the client will at some point disconnect since it's unable to get an IP (depending on what your client is)

What did you use to connect to the BLE peripheral? The BLE examples don't support pairing - you can connect but not pair

jsprog, 2024-10-16:
Sorry for rising this unnecessary issue, probably you'd move it to discussions instead.

> Please note the access point examples DON'T include a DHCP server - did you make sure to use a static IP? Otherwise, the station will most probably the client will at some point disconnect since it's unable to get an IP (depending on what your client is)

It's already stated in the examples to use a static IP address (done).

> What did you use to connect to the BLE peripheral? The BLE examples don't support pairing - you can connect but not pair

I didn't connect anything, I only tested the BLE examples after failing the WiFi access-point examples. Of course I tested with pairing and failed.

[Screencast 2024-10-16 18:10:14.webm](https://github.com/user-attachments/assets/e6a61be0-22a8-4021-8d70-aa502ad5f2c2)

jsprog, 2024-10-18:
@bjoernQ
Would you move this to discussions?

closed 2024-10-20 by jsprog

bjoernQ, 2024-10-21:
Thanks for closing this issue - I was unavailable during the past days but I think this issue is easier to find for anyone running into similar issues if we leave it as an issue

jsprog, 2024-10-21:
Well noted, usually I don't close issues too soon to allow for best decisions to repository's developers. At least, I should've waited for more days after the weekend.

Usually, I don't close the issues I rise too soon leaving time for developers