esp-rs/esp-hal #2362: i8080.set_byte_order seems to be incorrect starting 0.21.0

yanshay, 2024-10-16:
application using `i8080` stopped working starting `esp-hal 0.21.0`.
Detailed discussion regarding the issue:  #2352 .

Findings are this is due to change made in `set_byte_order` which the application relies on. 
In 0.21.0 the function change the byte ordering for both the 8bits and the 16bits case.
Prior to that the change was made depending on what the word size of the data is.
Seems like doing both changes in either case doesn't work well, at least for 8 bits i8080 and can't send the data properly to the device with if changing the byte order using `set_byte_order`.

0.20.0 code:
```
   pub fn set_byte_order(&mut self, byte_order: ByteOrder) -> &mut Self {
        let is_inverted = byte_order != ByteOrder::default();
        self.lcd_cam.lcd_user().modify(|_, w| {
            if size_of::<P::Word>() == 2 {
                w.lcd_byte_order().bit(is_inverted)
            } else {
                w.lcd_8bits_order().bit(is_inverted)
            }
        });
        self
    }
```

0.20.1 code:
```
    pub fn set_byte_order(&mut self, byte_order: ByteOrder) -> &mut Self {
        let is_inverted = byte_order != ByteOrder::default();
        self.lcd_cam.lcd_user().modify(|_, w| {
            w.lcd_byte_order().bit(is_inverted);               /// <- the line causing the issue in 8 bits case
            w.lcd_8bits_order().bit(is_inverted)
        });
        self
    }
```

MabezDev, 2024-11-07:
@Dominaezzz sorry for the ping, the milestone is near its end, do you still plan on fixing this, or shall I re-assign?

Dominaezzz, 2024-11-07:
Yes I'll take care of it

PR #2487 (2024-11-08): Fix I8080.set_byte_order()
## Thank you for your contribution!

We appreciate the time and effort you've put into this pull request.
To help us review it efficiently, please ensure you've gone through the following checklist:

### Submission Checklist 📝
- [x] I have updated existing examples or added new ones (if applicable).
- [x] I have used `cargo xtask fmt-packages` command to ensure that all changed code is formatted correctly.
- [x] My changes were added to the [`CHANGELOG.md`](https://github.com/esp-rs/esp-hal/blob/main/esp-hal/CHANGELOG.md) in the **_proper_** section.
- [x] I have add [... cut]