esp-rs/esp-hal #2364: Cannot start wifi on esp32s3

akesser, 2024-10-17:
## Bug description

I am trying to run the wifi examples on a esps3, but the code stops at
Device capabilities: Ok(EnumSet(Client))
Starting wifi

The chip is getting hot and can't be flashed after a reboot.

Thanks for your help in advance!

## To Reproduce

export SSID="myssid"
export PASSWORD="ChangeMe"
cargo xtask run-example esp-hal esp32s3 wifi_embassy_dhcp

## Expected behavior

The chip should be able to start the wifi and continue the script

## Environment

- Target device: esp32s3
- Crate name and version: esp-hal 0.21.1, main, 0.21.0

MabezDev, 2024-10-17:
It's probably working, it's most likely that you have the USB port plugged into the usb-serial-jtag port, which WiFi automatically disables for better RF performance. We have a feature to keep it enabled, could you try enabling `phy-enable-usb` feature and trying again?

akesser, 2024-10-17:
Thanks for the hint! It works now. I spend a pull request to add a note to all the wifi examples.