esp-rs/esp-hal #2370: Sigma-delta modulation driver

katyo, 2024-10-19:
## Motivations

Almost all ESP SoCs has Sigma-delta modulation periphery which can be used as a DAC (multi-channel).

## Solution

An SDM driver as a part of HAL.

## Alternatives

LEDC PWM driver also can be used in such case but SDM typically has lower noise.

## Links

[IDF HAL (C3)](https://github.com/espressif/esp-idf/blob/master/components/hal/esp32c3/include/hal/sdm_ll.h)
[IDF SDM interface](https://github.com/espressif/esp-idf/blob/master/components/esp_driver_sdm/include/driver/sdm.h)
[IDF SDM implementation](https://github.com/espressif/esp-idf/blob/master/components/esp_driver_sdm/src/sdm.c)
[IDF SDM example](https://github.com/espressif/esp-idf/blob/master/examples/peripherals/sigma_delta/sdm_dac/main/sdm_dac_example_main.c)

apps4uco, 2025-12-28:
Would it be a good idea to add the SDM peripheral to the support matrix on https://github.com/esp-rs/esp-hal/tree/main/esp-hal so that it is easier to find this issue, and also maybe give it a higher priority. Also, it would be excellent to support the cosine generation feature for motor control use cases.

ttzytt, 2026-05-24:
Hi all, I needed to use SDM for my current project. Unfortunately, it seems like this peripheral is not yet supported in esp-hal, so I spent some time researching the documentation and the source code of esp-idf. I was able to make SDM work on a C5 (the logic should be pretty much the same for other models) by direct register manipulation. Just sharing the code here in case anyone needs to use SDM before it's officially supported in esp-hal. 

The following code keeps increasing SDM density on GPIO0. 

```rust
#![no_std]
#![no_main]

use defmt::info;
use embassy_executor::Spawner;
use embassy_time::{Duration, Timer};
use esp_hal::clock::CpuClock;
use esp_hal::timer::timg::TimerGroup;
use panic_rtt_target as _;

esp_bootloader_esp_idf::esp_app_desc!();

const GPIO_EXT_BASE: usize = 0x6009_1e00;
const SIGMADELTA_MISC: usize = GPIO_EXT_BASE + 0x04;
const SIGMADELTA_CH0: usize = GPIO_EXT_BASE + 0x08;
const GPIO_SD0_OUT_IDX: u16 = 76;

unsafe fn reg_read(addr: usize) -> u32 {
    core::ptr::read_volatile(addr as *const u32)
}

unsafe fn reg_write(addr: usize, value: u32) {
    core::ptr::write_volatile(addr as *mut u32, value);
}

unsafe fn enable_sdm_clock() {
    let v = reg_read(SIGMADELTA_MISC);
    reg_write(SIGMADELTA_MISC, v | 1); // bit 0: SIGMADELTA_CLK_EN
}

unsafe fn set_sdm_ch0(density: i8, prescale_div: u16) {
    assert!((1..=256).contains(&prescale_div));

    let density_bits = density as u8 as u32;
    let prescale_bits = ((prescale_div as u32 - 1) & 0xff) << 8;
    let new = density_bits | prescale_bits;
    reg_write(SIGMADELTA_CH0, new);
}

#[esp_rtos::main]
async fn main(_spawner: Spawner) -> ! {
    rtt_target::rtt_init_defmt!();

    let config = esp_hal::Config::default().with_cpu_clock(CpuClock::max());
    let peripherals = esp_hal::init(config);

    let timg0 = TimerGroup::new(peripherals.TIMG0);
    let sw_interrupt =
        esp_hal::interrupt::software::SoftwareInterruptControl::new(peripherals.SW_INTERRUPT);
    esp_rtos::start(timg0.timer0, sw_interrupt.software_interrupt0);

    info!("Embassy initialized!");

    // configure GPIO0 as output
    let _sdm_pin = esp_hal::gpio::Output::new(
        peripherals.GPIO0,
        esp_hal::gpio::Level::Low,
        esp_hal::gpio::OutputConfig::default(),
    );

    // configure GPIO matrix to route SDM channel 0 output to GPIO0
    unsafe {
        esp_hal::peripherals::GPIO::regs()
            .func_out_sel_cfg(0) // to configure GPIO0
            .write(|w| {
                w.out_sel().bits(GPIO_SD0_OUT_IDX);  // select SDM channel 0 as the output signal
                w.inv_sel().bit(false);              // false -> signal not inverted
                w.oen_sel().bit(false);              // false -> output enable controlled by peripheral
                w.oen_inv_sel().bit(false)           // false -> output enable not inverted
            });
    }

    unsafe {
        enable_sdm_clock();
    }
    
    let mut density : i8 = -128;
    loop{
        Timer::after(Duration::from_millis(100)).await;
        unsafe{
            set_sdm_ch0(density, 4);
        }
        density = density.wrapping_add(1);
    }
}
```

Also, I’d be willing to help with implementing an SDM driver. I have quite some free time recently, and since I already spent a bit of time digging through the esp-idf implementation, I’d be happy to contribute. I may need some guidance on the right design/API direction and where to start, so I’d really appreciate any suggestions.

--- 

The following are some references: 

Definition of `GPIO_EXT_BASE` ([link to esp-idf source](https://github.com/espressif/esp-idf/blob/c1b7f3821282cf974ee6018f9787758cc7c05ffc/components/soc/esp32c5/register/soc/reg_base.h#L64))
> ```c 
> #define DR_REG_GPIO_EXT_BASE                      0x60091e00
> ```

Definition of `SIGMADELTA_MISC` ([link to esp-idf source](https://github.com/espressif/esp-idf/blob/c1b7f3821282cf974ee6018f9787758cc7c05ffc/components/soc/esp32c5/register/soc/gpio_ext_reg.h#L16-L22)): 

> ```c
> #define GPIO_EXT_SIGMADELTA_MISC_REG (DR_REG_GPIO_EXT_BASE + 0x4)
> /** GPIO_EXT_SIGMADELTA_CLK_EN : R/W; bitpos: [0]; default: 0;
> *  Configures whether or not to enable the clock for sigma delta modulation.
> *  0: Not enable
> *  1: Enable
>  *  %\label{fielddesc:GPIOSDSPISWAP}- [GPIOSD_SPI_SWAP] Reserved.
>  */
> ```

Definition of `SIGMADELTA_CH0` ([link to esp-idf source](https://github.com/espressif/esp-idf/blob/c1b7f3821282cf974ee6018f9787758cc7c05ffc/components/soc/esp32c5/register/soc/gpio_ext_reg.h#L31))

>```c 
> #define GPIO_EXT_SIGMADELTA0_REG (DR_REG_GPIO_EXT_BASE + 0x8)
>```

Definition of `GPIO_SD0_OUT_IDX` ([link to esp-idf source](https://github.com/espressif/esp-idf/blob/c1b7f3821282cf974ee6018f9787758cc7c05ffc/components/soc/esp32c5/include/soc/gpio_sig_map.h#L135))

>```c 
> #define GPIO_SD0_OUT_IDX                        76
>```

Fields of `SIGMADELTA_CH0` ([link to esp-idf source](https://github.com/espressif/esp-idf/blob/c1b7f3821282cf974ee6018f9787758cc7c05ffc/components/soc/esp32c5/register/soc/gpio_ext_struct.h#L31-L47))

> ```c
> /** Type of sigmadeltan register
> *  Duty cycle configuration register for SDM channel 0
>  */
> typedef union {
>   struct {
>        /** sdn_in : R/W; bitpos: [7:0]; default: 0;
>         *  Configures the duty cycle of sigma delta modulation output.
>         */
>        uint32_t sdn_in: 8;
>        /** sdn_prescale : R/W; bitpos: [15:8]; default: 255;
>         *  Configures the divider value to divide IO MUX operating clock.
>         */
>        uint32_t sdn_prescale: 8;
>        uint32_t reserved_16: 16;
>    };
>    uint32_t val;
> } gpio_ext_sigmadeltan_reg_t;
> ```

MabezDev, 2026-10-27:
Fixed by #5610, which adds the unstable `esp_hal::sdm` driver: https://github.com/esp-rs/esp-hal/commit/91e562452cf9b3a51d29c73d030c90144ad8c6a4

closed 2026-10-27 by MabezDev