esp-rs/esp-hal #2376: Explore using MCU boot as the second stage bootloader

MabezDev, 2024-10-21:
Related to #1973, we're leaning towards using MCUBOOT, but we need to explore if this is viable before we settle on it. The most important thing we need to know is can we change the ota slot addresses _without_ forcing the user to rebuild the MCU boot project.

jessebraham, 2024-10-24:
Don't want to count my chickens before they hatch, so being purposefully a bit cryptic here 😉 

Took me a few days to get going on this, unfortunately encountered a number of issues which slowed me down. However, based on some _very_ rudimentary tests I _think_ this should be viable.

I still have some work to do before I'm able to say for sure, but I'm hoping that I can have a better idea by end of day tomorrow. I will update this issue by early-to-mid next week with my findings.

jessebraham, 2024-11-04:
I'm able to build `mcuboot` and inject arbitrary values in for the required configuration values. These are stored in a contiguous block, and should be easily identifiable by tools e.g. `espflash`. Therefore, I believe this approach should be viable.

The next  steps are:

- Add the required linker scripts to support `mcuboot` in `esp-hal`
- Update our tooling (i.e. `espflash`) to support the `mcuboot` image format

jessebraham, 2024-11-07:
I guess this can probably be closed? The exploration is complete, additional work needs to be done but this involves writing new linker scripts in the HAL (which IMO is outside the scope of this issue) and updating tooling (which isn't in this repository, and has its own tracking issue already).

MabezDev, 2024-11-07:
I agree, but we still need to track the linker script work - we all know how fun they are :D. Could you open a new issue for that, and perhaps include some details on how to build MCUBOOT etc so whoever works on the linkerscripts can get a head start?

After that we can close this.

bjoernQ, 2024-11-07:
BTW here are the changes we had for supporting C3: https://github.com/esp-rs/esp-hal/pull/49/files#diff-f68012054272d7ce1611e965d7c6771a128c2e93fdcb4d8153cbcb9ef691062f

There is also setting up the MMU - I wonder if that was really necessary (I hope not) or not. If it is necessary, I guess ESP32 might add some fun to the task

jessebraham, 2024-11-07:
https://github.com/esp-rs/esp-hal/issues/2479

closed 2024-11-07 by jessebraham