esp-rs/esp-hal #2382: Improve release automation using xtask

jessebraham, 2024-10-22:
Currently, largely due to the sheer number of packages in this repository, releasing is a bit of a headache to put it mildly. Let's please discuss how we can automate this process to remove the possibility of human error as much as possible.

Some possible improvements:

- Determine which packages have changes, and why type of version increase the changes warrant (e.g. minor, patch)
- Automatically bump version numbers as needed, and update any dependent packages in the repository accordingly
- Automatically update `CHANGELOG.md` files prior to releasing
- Publish packages automagically

@MabezDev @tom-borcin let's please discuss this tomorrow in our meeting.

AnthonyGrondin, 2024-10-25:
My 2 cents;

I've been using the following tool: https://github.com/clog-tool/clog-lib for a while in one of my project and it's been going well. If this helps with the mentioned points above, that's good.

jessebraham, 2024-11-12:
I don't know if I'm just slow or what, but I have tried `cargo-semver-checks` multiple times now and failed to ever get it working, constantly getting new horrifying errors, so unfortunately at this point I'm going to rule this out as a possibility. If anybody else has had success using this for embedded libraries I would love to hear how you got it working.

I will see if I can try to come up with some alternative approach.

jessebraham, 2024-11-14:
I don't think the first item in the task list is reasonably possible at this point, this will still require manual review.

closed 2025-01-07 by jessebraham