esp-rs/esp-hal #2385: support embassy-net with simplified lifetimes/generics

jsprog, 2024-10-22:
## Motivations
`embassy-net` simplified lifetimes/generics since `be0d977 (Sep 16, 2024)`. Sooner or later, esp-hal need to catch up.
After updating embassy-net, I'm unable to create `Stack<'d>` instances due to a missing trait bound `Driver`.
```
the trait bound `WifiDevice<'_, WifiStaDevice>: Driver` is not satisfied`.
```

## Additional context
I already spotted some changes
- `struct Stack<D: Driver>` is now `Stack<'d>`
- `embassy_net::new(params)` instead of `embassy_net::Stack::new(params)`
- `embassy_net::new(params)` returns with a tuple `(Stack, Runner)`. `Stack` is for creation of `Sockets` (note: they are `Copy`) and `Runner` is for the background work.

bugadani, 2024-10-22:
> Sooner or later, esp-hal need to catch up.

esp-wifi will be updated once a new embassy-net version has been released. If you'd like to prepare an interim PR that brings our API closer, and maintain that PR until a release is made, we'd be grateful.

MabezDev, 2024-11-08:
I don't think we actually need to do anything here, `embassy-net-driver` didn't change. I've already updated a personal project to use git embassy-net + esp-wifi. Are you sure you enabled the embassy-net feature of esp-wifi? (Soon to disappear btw).

jsprog, 2024-11-16:
> I don't think we actually need to do anything here, `embassy-net-driver` didn't change. I've already updated a personal project to use git embassy-net + esp-wifi. Are you sure you enabled the embassy-net feature of esp-wifi? (Soon to disappear btw).

Did you read my text above about the simplified lifetimes/generics?

MabezDev, 2024-11-17:
> Did you read my text above about the simplified lifetimes/generics?

Did you read mine? I specifically mentioned `embassy-net-driver`, which hasn't changed. The only embassy-net changes required are in _your_ code, if you choose to use git embassy.

jsprog, 2024-11-18:
@MabezDev 

> I don't think we actually need to do anything here

`esp-wifi` need to catch up with with `embassy-net` and @bugadani already explained to wait for `embassy-net` release.

> Are you sure you enabled the embassy-net feature of esp-wifi? (Soon to disappear btw)

It's already enabled since a long time. Actually, I can't remember if it was the case more than a month ago, but I suppose It was enabled cause I was building with the release version of embassy-net with success!

> Did you read mine? I specifically mentioned `embassy-net-driver`, which hasn't changed.

Yes

> I've already updated a personal project to use git embassy-net + esp-wifi
>The only embassy-net changes required are in _your_ code, if you choose to use git embassy.

Let me guess:
- the `Driver` Trait didn't change
- `esp-wifi` implement the `Driver` trait for Wifi Interfaces
  - `pub struct WifiDevice<'d, MODE: WifiDeviceMode> {`
  - `impl<MODE: WifiDeviceMode> Driver for WifiDevice<'_, MODE> {`
  - `pub struct WifiStaDevice;`
  - `impl WifiDeviceMode for WifiStaDevice {`
- `embassy_net::Stack` has nothing to do with esp-wifi, but the `Driver` trait.

Then why the trait bound `WifiDevice<'_, WifiStaDevice>: Driver` is not satisfied`? It's quite strange!!! I'll try again to report later.

``` rust
// aff1e74 (before be0d977)
pub struct Stack<D: Driver> {
    pub(crate) socket: RefCell<SocketStack>,
    inner: RefCell<Inner<D>>,
}
impl<D: Driver> Stack<D> {
    /// Create a new network stack.
    pub fn new<const SOCK: usize>(
        mut device: D,
        config: Config,
        resources: &'static mut StackResources<SOCK>,
        random_seed: u64,
    ) -> Self {

// after be0d977
#[derive(Copy, Clone)]
pub struct Stack<'d> {
    inner: &'d RefCell<Inner>,
}
/// Create a new network stack.
pub fn new<'d, D: Driver, const SOCK: usize>(
    mut driver: D,
    config: Config,
    resources: &'d mut StackResources<SOCK>,
    random_seed: u64,
) -> (Stack<'d>, Runner<'d, D>) {
```

bugadani, 2024-11-18:
> and @bugadani already explained to wait for embassy-net release.

Don't listen to that guy, he doesn't know what he's talking about. esp-wifi doesn't depend on embassy-net, just embassy-net-driver.

Now, what is happening:
- You depend on a git version of embassy-net
- The git version of embassy-net depends on the git version of embassy-net-driver
- esp-wifi depends on the `crates.io` version of embassy-net-driver
- These are different crates, so rust will say that esp-wifi doesn't implement the interface of the git driver, because, well, it doesn't.

closed 2025-01-10 by bugadani