esp-rs/esp-hal #2398: Default CPU config of max instead of boot defaults?

MabezDev, 2024-10-23:
Discussion here on matrix: https://matrix.to/#/!LdaNPfUfvefOLewEIM:matrix.org/$RQ5btwvUWV83omJJZk1uAHf3eeFXdajv9vYtcqwFcB0?via=matrix.org&via=tchncs.de&via=envs.net

TLDR: it's probably more useful to configure the clocks to max, than leave them at boot defaults.

Dominaezzz, 2024-10-23:
There's also the option of doing this in the template instead of the hal.

bugadani, 2024-10-24:
I think making the template set max speed is a good idea. It makes it obvious that such an option exists, and it's also not a breaking change.

bjoernQ, 2024-10-24:
+1 - additionally we could also change the example on the docs landing page (blinky) to configure max CPU speed (and have a comment there) .... just in case someone reads the documentation 😄

closed 2024-10-24 by MabezDev