esp-rs/esp-hal #2408: I2C timeouts on Adafruit ESP32-C6 Feather

blinry, 2024-10-25:
## Bug description

When trying to use I2C on the Adafruit ESP32-C6 Feather, I always run into timeouts.

The example below tries to communicate with the built-in MAX17048 li-po battery level sensor, but this also happened when trying to talk to a Senseair Sunrise CO2 sensor (which worked on a Adafruit ESP32 V2 Feather, using similar code).

## To Reproduce

1. Clone https://github.com/blinry/esp32-c6-i2c-test
2. Attach a Adafruit ESP32-C6 Feather
3. Run `cargo run --release`

Result:

```====================== PANIC ======================
panicked at /home/blinry/.cargo/registry/src/index.crates.io-6f17d22bba15001f/max17048-0.1.0/src/lib.rs:26:41:
called `Result::unwrap()` on an `Err` value: TimeOut
```

I reproduced this on the main branch.

Note: I'm using the max17048 crate for convenience, but also got these issues when attempting a `write_read` call myself.

## Expected behavior

I2C communication with the MAX17048 works when using the Arduino IDE, so I'm assuming the problem is not my hardware. This [Arduino pin definition](https://github.com/espressif/arduino-esp32/blob/master/variants/adafruit_feather_esp32c6/pins_arduino.h) confirms that SDA is pin 19, and SCL is pin 18.

## Environment

- Target device: Adafruit ESP32-C6 Feather
- Crate name and version: esp-hal main

Let me know if there's something I can do to help debug this further!

blinry, 2024-10-25:
I tried measuring the activity on the SDA/SCL pins.

1: In an Arduino sketch (the example that comes with Adafruit's MAX17048 library):

[image]

(Looks reasonable to me, and the example works.)

2: Using my minimal Rust example in [this repo](https://github.com/blinry/esp32-c6-i2c-test):

[image]

(Pins seem to go high for around 0.5 s, until the panic. No communication visible at all.)

blinry, 2024-10-25:
I confirmed that I'm able to write to the pins (again, my assumption is SDA = gpio19 and SCL = gpio18) if I make them into `Outputs`, and `toggle()` them.

But putting them into an `I2c`, and doing a `i2c.write(0x36, &[0x42]);` seems to produce no pin activity, even though I think it should.

What could be going wrong here?

bugadani, 2024-10-25:
Which exact esp-hal version are you using?

blinry, 2024-10-25:
Right now the latest commit on the main branch, 6bf03f63a9bb6ee558913fcef5392a3048e91b5b.

bugadani, 2024-10-25:
I just quickly slapped the latest commit onto my toy device that has an I2C fuel gauge (a MAX17055 fwiw) and at least our async I2C isn't completely broken. I'll look into this issue, probably tomorrow but early next week at the latest.

bugadani, 2024-10-28:
Unfortunately, the sample project you provided seems to work as expected for me - with the slight caveat that I'm using different hardware. I don't have your sensor, so just running it on my devkit (an ESP32-C6-DevKitC-1), connected to nothing, I see the following waveform and the expected `AckCheckFailed` error:

[image]

I see the same error on a custom board (that, by a happy incident, also uses 19/18 as SDA/SCL) trying to read a different sensor, which also seems legit to me.

Can you please make sure that I2C_PWR (IO20) is high before attempting the first I2C operation?

blinry, 2024-10-28:
Setting IO20 high solved the issue! Would never have thought of that.

Thank you so much! Keep up the good work, I really like the refactorings you've done in the last releases! \o/

closed 2024-10-28 by blinry