esp-rs/esp-hal #2412: Accessing WifiEvent data

Easyoakland, 2024-10-26:
## Motivations

Getting the data from `WifiEvents` for application. I'm particularly interested in accessing `wifi_event_ap_staconnected_t` and `wifi_event_ap_stadisconnected_t ` data to maintain a list of connected and disconnected clients and to do something based on the mac of the connecting or disconnecting client.

From what I can tell, this data is made available to the `event_post` function [here](https://github.com/esp-rs/esp-hal/blob/6bf03f63a9bb6ee558913fcef5392a3048e91b5b/esp-wifi/src/wifi/os_adapter.rs#L853) but is discarded.

## Solution

A way to access the data from `WifiEvents`. If not a blocking callback, then the events would have to buffer since losing an event matters.

## Alternatives

For my particular problem of maintaining a list of connected macs, I can somewhat use the `sta_list` functions as [here](https://github.com/Easyoakland/esp-hal/tree/sta-list) (aside: should I submit a pull request?) but the list appears to sometimes be out of sync and I have to manually compare the list before and after an event to find which entry was added or removed. It would make more sense to use the data of the event if it were available.

## Additional context

Easyoakland, 2024-10-26:
Also, does `esp-hal` support setting up event callbacks at all? As best I can tell, the only supported way to do this is using the `async` `WifiController::wait_for_*` functions in a super-function which has a unique reference to the controller dispatches to callback by matching the received event.

bjoernQ, 2024-10-28:
Having a way for user code to 

a) subscribe to events and/or dequeue events
b) get event data (see https://github.com/espressif/esp-idf/blob/9106c43accd9f5e75379f62f12597677213f5023/components/esp_wifi/include/esp_wifi_types_generic.h#L1032-L1146)

Seems like a good idea. (we don't have it yet)

Ideally the user could mask out events they are not interested in

Easyoakland, 2024-10-31:
Would an intrusive doubly-linked-list of `Fn(event_type)` work? [the c implementation](https://github.com/espressif/esp-idf/blob/v5.3.1/components/esp_event/esp_event.c) appears to allocate a singly linked list, but I don't think allocs are needed (especially in the common case of setting all handlers at the beginning of main).

Thoughts? Would adding `Pin` to `esp-hal`'s api be an issue?

Easyoakland, 2024-10-31:
Also, would there be any issues adding a new dependency for this?

bjoernQ, 2024-10-31:
In general, `alloc` wouldn't be a problem since we already require alloc - but since it's pushing data from an "internal" callback the memory should be allocated in internal-RAM (not PSRAM) and the implementation shouldn't allocate on its own.

We already have a naive `RawQueue` internally used by the wifi / ble - drivers: https://github.com/esp-rs/esp-hal/blob/25d8390d68f5b996c3df5200b485ef217cf5793b/esp-wifi/src/compat/common.rs#L30 and there will be more once #2396 lands

Using `Pin` in the public API is not a problem if it's inevitable but IIRC the events are not "huge" and copying them should be fine - so maybe no need to have `Pin` in public API?

Regarding adding dependencies: depends on which dependency I guess

Easyoakland, 2024-10-31:
> In general, `alloc` wouldn't be a problem since we already require alloc - but since it's pushing data from an "internal" callback the memory should be allocated in internal-RAM (not PSRAM) and the implementation shouldn't allocate on its own.

I'm not sure what you mean by the implementation shouldn't allocate on its own.

> We already have a naive `RawQueue` internally used by the wifi / ble - drivers:

I'll have to look at that. Any reason you use that instead of an external implementation?

> https://github.com/esp-rs/esp-hal/blob/25d8390d68f5b996c3df5200b485ef217cf5793b/esp-wifi/src/compat/common.rs#L30
> and there will be more once #2396 lands

More what? Heap allocations?

> Using `Pin` in the public API is not a problem if it's inevitable but IIRC the events are not "huge" and copying them should be fine - so maybe no need to have `Pin` in public API?

I wasn't going to pin the events.
I was more thinking something like this:
```rs
register_handler::<Event, F: Fn(&Event)>(handler: Pin<&mut EventHandlerUnregistered<F>>) -> EventHandlerRegistered<'_, F>
```
would add the function to the linked list where
```rs
struct EventHandlerUnregistered<'a, F> {
    f: F,
    next: ptr,
    back: ptr,
}
struct EventHandlerRegistered<'a, F> {
   on_drop: &'a mut EventHandlerUnregistered
}
impl Drop for EventHandlerRegistered {
fn drop(){
// Remove from linkedlist after acquiring the global rwlock
}
}
```

There would a linked-list wrapped in a fair rwlock (for deregistration) per event type with the starting pointer in some array: `handlers: [EventTy1StartPtr, EventTy2StartPtr, ...]` 

and a new handler would be added like:
```rs
let unreg = EventHandlerUnregisted::new(f);
let unreg_pin = pin!(unreg); // or use Box::pin
let reg = register_handler::<Event1, _>(unreg_pin);
```

Easyoakland, 2024-10-31:
If there's not already a preferred rwlock implementation I was going to use https://crates.io/crates/maitake-sync. I was also going to look at https://crates.io/crates/cordyceps/0.3.2 and https://crates.io/crates/pin-list for the linked list implementation before writing it from scratch.

bjoernQ, 2024-10-31:
> I'm not sure what you mean by the implementation shouldn't allocate on its own.

e.g. a `Vec` exceeding its capacity would re-alloc

Thinking about it: Is this use-case common enough to justify maintaining a queue of events? Wouldn't it be enough to offer an API to register a (blocking) callback and the user can decide what they really want to do with the events? 

(You already mentioned that initially and I tend to be in favor of that)

Easyoakland, 2024-10-31:
Its not a queue of events. Its a linked list of event handlers (blocking functions). The pinning is so the reference to the handler is stable.

bjoernQ, 2024-10-31:
> It's not a queue of events. Its a linked list of event handlers (blocking functions). The pinning is so the reference to the handler is stable.

What would be the use-case for registering an arbitrary number of callbacks?

Easyoakland, 2024-10-31:
1. Its implemented for the `c` version and I assumed it would be wanted here as well.
2. It enables decoupling when you have multiple callbacks which should respond to the event that aren't related.
The alternative is a "super-callback" which calls all the others, but this is very tedious if using a library or trying to abstract code because you get (for each event):
```
fn super_callback(event: Event) {
    callback1(&event)
    callback2(&event)
    ...
}
```
 It gets worse if the other callbacks need setup code.
3. It enables storing state with the callbacks without having to use statics because the linked list is of `dyn Fn` and not `fn`. This wouldn't work for a single callback, so you would need to use statics, which is unfortunate.

bjoernQ, 2024-10-31:
I'd say let's start with a single callback and keep things simple for now. Especially since other APIs are like that and if we go this way here, we should have all of the API in sync

MabezDev, 2024-10-31:
Before we commit to this callback approach, I think we should consider coupling the `WifiEvent` enum _with_ the data.

I.e

`WifiEvent::ScanDone(Box<wifi_event_sta_scan_done_t>)`, or something like that might be a more rusty approach.

Easyoakland, 2024-10-31:
> Before we commit to this callback approach, I think we should consider coupling the `WifiEvent` enum _with_ the data.
> 
> I.e
> 
> `WifiEvent::ScanDone(Box<wifi_event_sta_scan_done_t>)`, or something like that might be a more rusty approach.

I don't know how that would work.
1. Boxing the event data doesn't do anything since the event data is already `Sized`.
2. Where would the events be stored? Currently they're stored in a static which has capacity 1, but that only works because the events are identical, so losing one doesn't mean anything.
3. If we're going to argue what is most "rusty" the standard library handles panics (which are arguably like events) with [set_hook](https://doc.rust-lang.org/std/panic/fn.set_hook.html) which is a linked-list of [Box](https://doc.rust-lang.org/std/boxed/struct.Box.html)<dyn [Fn](https://doc.rust-lang.org/std/ops/trait.Fn.html)(&[PanicHookInfo](https://doc.rust-lang.org/std/panic/struct.PanicHookInfo.html)<'_>) + [Sync](https://doc.rust-lang.org/std/marker/trait.Sync.html) + [Send](https://doc.rust-lang.org/std/marker/trait.Send.html) + 'static> where items can only be added, not removed.

Easyoakland, 2024-11-03:
By the way, I get a linker error trying to build the example on the latest commit, even before changing anything, and had to fix it with:
```diff
- esp-wifi-sys = { version = "0.6.0", git = "https://github.com/esp-rs/esp-wifi-sys.git", rev = "a3e2ed64e9095f120bbcebe6287ddf62760774db"}
+ esp-wifi-sys = { version = "0.6.0"}
```
Error:
```
[...]/.cargo/target\xtensa-esp32-none-elf\release\build\esp-wifi-sys-5deb9f734c403436\out\libbtdm_app.a: file not recognized: file format not recognized 
          collect2.exe: error: ld returned 1 exit status
```

closed 2024-11-13 by jessebraham