esp-rs/esp-hal #2413: Spi does not implement SpiDevice?

MatthijsReyers, 2024-10-26:
## Bug description

I'm trying to write to an SD card with the `embedded-sdmmc` crate (latest verion: `0.8.0`), which requires a struct that implements the `SpiDevice<u8>` trait. Now the documentation of `/esp-hal/src/spi/master.rs` mentions the `SpiDevice` trait, but after grepping this repository I found that this doc string is in fact the only instance of the word `SpiDevice` ever appearing in the `esp-hal` repo. Should the `Spi` struct not implement the `SpiDevice` trait? 

I have used the `embedded-sdmmc` before with an earlier version of `esp-hal` and this worked without a problem back then.

## To Reproduce

For reference, this is the code that I would expect to work:

```rs
use esp_backtrace as _;
use esp_hal::{delay::Delay, gpio::Io, prelude::*, spi::{master::Spi, SpiMode}};
use embedded_sdmmc::SdCard;

#[entry]
fn main() -> ! {
    let peripherals = esp_hal::init(esp_hal::Config::default());
    let io = Io::new(peripherals.GPIO, peripherals.IO_MUX);

    let cs = io.pins.gpio7;
    let sck = io.pins.gpio8;
    let miso = io.pins.gpio9;
    let mosi = io.pins.gpio10;
    let spi = Spi::new(peripherals.SPI2, 20_000u32.kHz(), SpiMode::Mode0).with_pins(sck, mosi, miso, cs);

    let delay = Delay::new();
    let sdcard = SdCard::new(spi, delay);

    loop {}
}
```

## Expected behavior

## Environment

- Target device: [e.g. ESP32-S3] 
- Crate name and version: [e.g. esp-hal 0.20.0]

Dominaezzz, 2024-10-26:
You want [ExclusiveDevice](https://github.com/rust-embedded/embedded-hal/blob/master/embedded-hal-bus/src/spi/exclusive.rs#L19)

Dominaezzz, 2024-10-26:
See also https://github.com/esp-rs/esp-hal/blob/v0.21.1/esp-hal/src/spi/master.rs#L21

closed 2024-10-26 by Dominaezzz