esp-rs/esp-hal #2416: Uart API improvements

MabezDev, 2024-10-27:
* `Uart::new()` should be infallible, only applying some invalid config should be invalid
* Remove the independent `change_x` mutator methods in favour of `apply_config(&mut self, config: &Config)`

bugadani, 2024-10-28:
Should all other drivers be reworked to use a config struct? My main worry is that clock speed calculations on SPI aren't trivial and changing any configuration means changing everything with this approach.

Should we keep `with_config` that moves `self`, along with `apply_config`?

bugadani, 2024-10-28:
Considering https://github.com/esp-rs/esp-hal/issues/2389 and the cost of a full reconfig, do we even want config structs? What if we got rid of them in favour of single option setters?

MabezDev, 2024-10-28:
> Considering https://github.com/esp-rs/esp-hal/issues/2389 and the cost of a full reconfig, do we even want config structs? What if we got rid of them in favour of single option setters?

Hmm, I quite like the idea of a config struct because it neatly packages everything in one place. In the case of Uart for example, we have to forward methods from `UartRx` to `Uart` and the same for `UartTx`. All three could just have one `apply_config` method and each will update there respective parts of the config.

Regarding the cost, we can actually be quite smart about this in the future. We can keep a copy of the current config and then only update the options that have changed. If we didn't want to pay the memory cost of storing the config in the driver, we could also just check the registers themselves (though this wouldn't work for registers that require some calculation from config inputs).

bugadani, 2024-10-28:
Checking the registers before writing the same or different value into them doesn't make much sense :)

Dominaezzz, 2024-10-28:
> My main worry is that clock speed calculations on SPI aren't trivial

There's a solution for this one. https://github.com/esp-rs/esp-hal/issues/1666#issuecomment-2154512074

bugadani, 2024-11-02:
I'd like to batch https://github.com/esp-rs/esp-hal/issues/1919 into this issue. Not necessarily for UART, but mainly for SPI - we get per-device configuration for free using https://docs.embassy.dev/embassy-embedded-hal/git/default/shared_bus/asynch/spi/struct.SpiDeviceWithConfig.html

bugadani, 2024-11-05:
> Uart::new() should be infallible, only applying some invalid config should be invalid

We have a few options on how to apply initial configuration, but we'll have to pick which one we'll do:

1. `new() -> Driver;` and `new_with_config(config) -> Result<Driver, Error>` - default config is rarely what the user needs, we don't have to optimize for that
2. Only have `new() -> Driver` and `apply_config` - forces an extra call, which is probably fine for SPI shared bus but not really good elsewhere
3. `new(config) -> Driver`. We could panic if the driver is initialized with an invalid config as it's likely the user will just unwrap it and get the panic anyway.
4. `new(config) -> Result<...>` - probably the least offensive option. The compiler is often smart enough to remove the panic, but `new().unwrap()` isn't terribly elegant.

bjoernQ, 2024-11-06:
My gut feeling is 4 but my common sense would prefer 3 

I assume that people will use fixed configurations so they will see the panic early in development and not have devices crash randomly in the field weeks later

bugadani, 2024-11-06:
I guess if we choose 3 and someone absolutely needs the driver to not panic but provide an error, they can create the driver with `Config::default()` and call `apply_config` with custom configuration. I expect in case bus configuration isn't known at build time (is read from some other source, like persistent storage) this might see some use. I consider this a special enough case to not actively optimize for (which would be option 4).

JurajSadel, 2024-11-06:
I like the 3 the most.

Dominaezzz, 2024-11-06:
I also like 3 most.

I sorta like 4 but I'd prefer if the error handling happened before creating the driver. (i.e. make it impossible to create an invalid config) which just makes it look like 3 anyway.

bugadani, 2024-11-06:
> (i.e. make it impossible to create an invalid config)

Most of the difficulty comes from baud rates, especially where the clock source is configurable. In that case, it is unfortunately rather difficult to check a config in compile time.

Dominaezzz, 2024-11-07:
I mean something along the lines of `Config::set_frequency` returning a `Result`.

MabezDev, 2024-11-07:
I'm leaning towards 3, but I'm wondering about the case where driver initialization might happen later in the program. I see the value in panicking early if it's initialized early, but is it acceptable to panic later?

Another consideration, what if the config is stored on an eeprom at runtime, the next reboot the settings are loaded from there. If someone somehow stored an invalid config we'll essentially bootloop. The solution to this in this case is to always use apply and pass `Default::default` into the constructor, but it's worth thinking about.

MabezDev, 2024-11-07:
Quoting our API-GUIDELINES

> Prefer compile-time checks over runtime checks where possible, prefer a fallible API over panics.

I feel like this pushes us closer to 4 or 2, unless we want to rethink this guideline.

MabezDev, 2024-11-22:
Whilst 3 is my favorite ergonomically, I think to support a wide range of use cases we should go with option 4.

Option 3 relies to heavily on the fact a driver is created at the beginning of the program and lives the entire time. This doesn't play nicely with the peripheral pattern for example, and may end up with unexpected panics.

Option 3 also pushes users to emulate option 2 for the case where they don't want to deal with those panics, meaning they call with default config and use `apply_config` (applies config twice, not great.)

> I sorta like 4 but I'd prefer if the error handling happened before creating the driver. (i.e. make it impossible to create an invalid config) which just makes it look like 3 anyway.

This would be nice, but I don't think it's possible to implement this for every driver, as it can depend on a lot of variables outside of what the config can see. One example would be to choosing a clock source that's not enabled, should the config be checking this and returning an error, or the driver? I would argue a config shouldn't be poking around in clock registers.

I think if we go with option 4 now, we can come up with some more ergonomic helpers to alleviate the pain of `unwrap` later down the line.

PR #2610 (2024-11-28): Constructor consistency update
## Thank you for your contribution!

We appreciate the time and effort you've put into this pull request.
To help us review it efficiently, please ensure you've gone through the following checklist:

### Submission Checklist 📝
- [x] I have updated existing examples or added new ones (if applicable).
- [x] I have used `cargo xtask fmt-packages` command to ensure that all changed code is formatted correctly.
- [x] My changes were added to the [`CHANGELOG.md`](https://github.com/esp-rs/esp-hal/blob/main/esp-hal/CHANGELOG.md) in the **_proper_** section.
- [x] I have [... cut]
closed 2024-11-28 by MabezDev