esp-rs/esp-hal #2428: ESP32-C6 async Executor not sleeping

kenohassler, 2024-10-29:
## Bug description

When yielding from an async task, the power consumption stays constant (~30mA). From the descriptions on the docs, I would have thought that the executor puts the ESP to sleep while there is nothing to do, but apparently, this does not happen.
A minimal example that calls `rtc.sleep_light()` works however, it reduces consumption to 6mA.
Now this might be a misunderstanding, maybe I am supposed to manually sleep and wake the chip, but that doesn't sound correct to me since that is basically equivalent to writing my own async executor (right?).
Given that the sleep support for the C6 was added only recently, is this maybe just a missing feature? I'm happy for any hint where to look next :)

## To Reproduce

Measure the power consumption while running this code:

```rust
#[embassy_executor::task]
async fn run(io: Io) {
    let mut led = Output::new(io.pins.gpio7, Level::High);
    loop {
        led.toggle();
        Timer::after(Duration::from_millis(5_000)).await;
    }
}

#[esp_hal_embassy::main]
async fn main(spawner: Spawner) {
    esp_println::logger::init_logger_from_env();
    let peripherals = esp_hal::init(esp_hal::Config::default());

    let timg0 = TimerGroup::new(peripherals.TIMG0);
    esp_hal_embassy::init(timg0.timer0);

    let io = Io::new(peripherals.GPIO, peripherals.IO_MUX);
    spawner.spawn(run(io)).ok();
}
```

I see a power draw around 30mA with the LED off and 32mA with the LED on.

## Expected behavior

I would have expected power draw to decrease significantly between the wakes to a level comparable to the 6mA I measured when manuall calling `sleep_light()`.

## Environment

- Target device: ESP32-C6
- Crate name and version: esp-hal 0.21.1, esp-hal-embassy 0.4.0

bugadani, 2024-10-29:
Thanks for noticing this, the executor is not meant to sleep currently.

closed 2024-10-29 by bugadani

kenohassler, 2024-10-30:
Hi @bugadani, thanks for the prompt answer. May I ask what is the best way forward if I want to write a low-power application? I thought of sleeping in an async task, but one would have to "guess" a good duration then? Is there an obvious choice, or should I basically use a different chip for that?

bugadani, 2024-10-30:
You can try copy-pasting the executor implementation, add the light sleep call and see if it's still working for you. We're not doing this because it takes quite a bit of setup to enable light sleep and low power applications aren't our focus at this moment, but someone experimenting in this area would be helpful.