esp-rs/esp-hal #2432: esp-hal-embassy clock initialization with SysTimer Alarm panics shortly after

lulingar, 2024-10-30:
## Bug description

With esp-hal `v0.21.1`, initializing embassy with **a single one** of the SysTimer alarms results in a runtime panic (different than the one in #2431), shortly after `esp_hal_embassy::init(...);`

## To Reproduce

1. With the following program:

```
#[main]
async fn main(spawner: Spawner) -> ! {
    esp_println::logger::init_logger_from_env();
    let peripherals = esp_hal::init(esp_hal::Config::default());

    let systimer = SystemTimer::new(peripherals.SYSTIMER).split::<Target>();
    esp_hal_embassy::init(systimer.alarm0);

    let mut cpu_control = CpuControl::new(peripherals.CPU_CTRL);
    let io = Io::new(peripherals.GPIO, peripherals.IO_MUX);

    // WiFi PHY start
    let timg0 = TimerGroup::new(peripherals.TIMG0);
    let mut rng = Rng::new(peripherals.RNG);
    esp_alloc::heap_allocator!(74 * 1024);
    let seed = ((rng.random() as u64) << 32) | (rng.random() as u64);
    let init = esp_wifi::init(
        esp_wifi::EspWifiInitFor::Wifi,
        timg0.timer0,
        rng,
        peripherals.RADIO_CLK,
    )
    .unwrap();
    let (wifi_interface, controller) =
        esp_wifi::wifi::new_with_mode(&init, peripherals.WIFI, WifiStaDevice).unwrap();
    ...
}
```

2. and the following dependencies in `Cargo.toml`:

```
esp-hal = { version = "0.21.1", features = ["esp32s3", "log"] }
esp-alloc = { version = "0.5.0", features = ["nightly"] }
esp-hal-embassy = { version = "0.4.0", features = [
    "esp32s3", "log",
    "integrated-timers"
] }
embassy-executor     = { version = "0.6.1", features = [
    "executor-interrupt", "executor-thread",
    "integrated-timers",
    "log", "task-arena-size-16384"
] }
static_cell = { version = "2.1.0", features = ["nightly"] }
...
```

3. I get a runtime panic:

```
====================== PANIC ======================
panicked at /home/being/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-hal-embassy-0.4.0/src/time_driver.rs:167:30:
index out of bounds: the len is 1 but the index is 1
```

4. [Observation](https://matrix.to/#/!LdaNPfUfvefOLewEIM:matrix.org/$d05kMz-RY8WPH84fD9kH26HroqQxMbg7Y6UA3_lMdks?via=matrix.org&via=tchncs.de&via=envs.net) from @bugadani: Using the other timers instead, works (i.e. no panic in over 1 hour):

```
 let timg0 = TimerGroup::new(peripherals.TIMG0);
 let timer1: AnyTimer = timg0.timer1.into();
 let timg1 = TimerGroup::new(peripherals.TIMG1);
 let timer10: AnyTimer = timg1.timer0.into();
 esp_hal_embassy::init([timer1, timer10]);
```

## Expected behavior

There should be no panic!

## Environment

- Target device: ESP32-S3
- Crate name and version: esp-hal v0.21.1
- `rustc --version`: `rustc 1.82.0-nightly (ffd10cecc 2024-10-25) (1.82.0.1)`
- `espup --version`: `espup 0.11.0`

bugadani, 2024-10-30:
This is not an issue (not with the code, at least), you need to provide as many alarms as you create executors.

lulingar, 2024-10-30:
Oh I didn't know about that requirement, thank you! I'll close this report.

PR #2451 (2024-11-04): Systimer improvements
- Optimize register access
- Clean up duplicate code - this might be enough to prevent the panics seen in #2431
- Improve panic message for #2432
- Ensure consistency when reading the counter.

Closes #2431
Closes #2432

At least in theory, I did not spend time reproducing the race issue. I do not think we can test the race condition easily so I didn't add a test case.
closed 2024-11-04 by bugadani