esp-rs/esp-hal #2440: Debug and re-enable I2S tests on ESP32

bugadani, 2024-10-31:
Disabled in #2439 due to noise.

bugadani, 2025-01-13:
A (probably separate) ESP32 I2S issue can be reproduced by https://github.com/liebman/esp-hub75/commit/e5f83929d527bbc716b5dffe3c9b6c508ab84f64

Placing an `info` between the following lines makes the example work:

https://github.com/liebman/esp-hub75/blob/e5f83929d527bbc716b5dffe3c9b6c508ab84f64/examples/i2s_parallel.rs#L260-L264

This issue feels very similar to other i2s/spi issues (like https://github.com/esp-rs/esp-hal/issues/2909) where we have to wait for some amount of time, otherwise an async operation hangs.

liebman, 2025-01-19:
> A (probably separate) ESP32 I2S issue can be reproduced by [liebman/esp-hub75@e5f8392](https://github.com/liebman/esp-hub75/commit/e5f83929d527bbc716b5dffe3c9b6c508ab84f64)
> 
> Placing an `info` between the following lines makes the example work:
> 
> https://github.com/liebman/esp-hub75/blob/e5f83929d527bbc716b5dffe3c9b6c508ab84f64/examples/i2s_parallel.rs#L260-L264
> 
> This issue feels very similar to other i2s/spi issues (like [#2909](https://github.com/esp-rs/esp-hal/issues/2909)) where we have to wait for some amount of time, otherwise an async operation hangs.

This also happens on the `esp32s3` using lcd_cam - example [here](https://github.com/liebman/esp-hub75/blob/e5f83929d527bbc716b5dffe3c9b6c508ab84f64/examples/lcd_cam.rs).

On the `esp32c6`, however it works as expected - example [here](https://github.com/liebman/esp-hub75/blob/e5f83929d527bbc716b5dffe3c9b6c508ab84f64/examples/parl_io.rs).

All the above examples are very similar.

Seems like something with the DMA interrupts on xtensa not playing nice with embassy executor.

Dominaezzz, 2025-01-19:
The I8080 driver doesn't use the DMA interrupts fwiw, it has it's own "I'm done" interrupt.

liebman, 2025-01-20:
More on the i8080 - it never sends anything - not even one clock.

bugadani, 2025-01-21:
Thank you both, a number of problems have been discovered around DMA and the LCD_CAM peripheral, but neither of those are actually related to this particular issue.