esp-rs/esp-hal #2443: Rng should rate limit to avoid draining entropy before it is added by the hardware

MabezDev, 2024-10-31:
See https://github.com/espressif/esp-idf/blob/9106c43accd9f5e75379f62f12597677213f5023/components/esp_hw_support/hw_random.c#L46-L88

bugadani, 2024-11-05:
Do we have some better information about how the RNG works, than the TRM? The TRM implies that the peripheral is always TRNG, but we're using it as though it acted as a PRNG when it doesn't have enough entropy.

Porting the C code wouldn't be too hard, although I have some worries about multi-core concurrent access.

bugadani, 2024-11-06:
> If none of the above conditions are true, the output of the RNG should be considered pseudo-random only.

I would be more comfortable seeing this in a TRM, but okay 😅