esp-rs/esp-hal #2447: Allow the `esp-wifi` stack to run on the 2nd core

ProfFan, 2024-11-01:
## Motivations

This would allow for more predictable timings for real-time tasks on the 1st core (as it is the default)

## Solution

For example, the timer interrupt code should be configurable to run on the APP_CPU

## Alternatives

Let the user move everything to spawn on an executor on the APP_CPU and keep `esp-wifi` running on PRO_CPU

## Additional context

N/A

DBLouis, 2025-02-13:
Is it really not possible to run the wifi stack on the second core? It should be mentioned somewhere

MabezDev, 2025-02-13:
It's possible it just doesn't work at the moment, it needs someone to debug it.

bjoernQ, 2025-02-13:
Currently ProCore/core0 are hardcoded in a few places. 

Interestingly wifi seems to work for me on the second core anyways on ESP32/S3
On S3 even BLE works for me on the second core but on ESP32 it doesn't (and even changing the hardcoded core0 doesn't make it work)

MabezDev, 2025-08-27:
Example of such hard-coding: https://github.com/esp-rs/esp-hal/blob/e42f7d2e06eac20afe0a0762677ebe9acd8043fe/esp-radio/src/wifi/os_adapter/esp32.rs#L34

jessebraham, 2025-09-02:
Based on my testing, both the `ble/scanner` and `wifi/dhcp` examples seem to run correctly on either core for both the ESP32 and the ESP32-S3. It also seems that the interrupts are being serviced by the correct core without any further changes required. Maybe somebody else wants to confirm this, but as far as I can tell this issue can just be closed.

PR #4037 (2025-09-05): Add a note regarding running on the second core to `esp-radio` documentation
See title. Feel free to nitpick, wasn't entirely sure what to put, I think I covered the major points at least, but happy to make any further changes.

Closes #2447