esp-rs/esp-hal #2493: Stable driver analysis: I2C master

MabezDev, 2024-11-08:
As part of #2491

bjoernQ, 2024-11-08:
We currently don't support 10-bit addressing mode (while the HW supports it and eh-traits define it)

jessebraham, 2024-12-03:
Just braindumping in this comment for now, will be edited many times I'm sure. Possibly some of these things are supported, or out of scope. Will convert things to issues as needed at a later date.

Missing functionality:

- I2C slave mode
- Digital noise filtering
- Dual-address mode
- Serial/parallel data converter
- Clock stretching
- SDA/SCL configuration (need to expand on the details here)
- Timeout control
- FIFO/Non-FIFO mode
- Support for all hardware interrupts

bugadani, 2024-12-03:
> Dual-address mode

This is a slave-mode feature as far as I can tell, please double check.

bjoernQ, 2024-12-09:
> I2C_CHUNK_SIZE needs to be clarified - all devices have 32x8bit RAM

ESP32-C2 is 16 × 8 bits

IIRC correctly it's only relevant for ESP32 / ESP32-S2 - on later chips it's arbitrary value. Probably because on ESP32/ESP32-S2 there is no FIFO_WM

bjoernQ, 2024-12-10:
Nitpick and not important for stabilization: `setup_write` and `setup_read` could use `impl Trait` for `cmd_iterator` to better align with other parts of the code