esp-rs/esp-hal #2498: Allow docs to show unstable features

MabezDev, 2024-11-08:
We could simply do this with cfg(feature = "unstable"), but I think we can do better by using [crates.io/crates/instability](https://crates.io/crates/instability). We will more than likely want to PR (or fork) that crate to add #![feature(doc_cfg)] support so that our unstable APIs are still discoverable.

We don't have to use instability, and could roll our own but I think one thing that is non-negotiable is having doc_cfg support, else our docs will be pretty sparse.

bugadani, 2024-11-21:
With the `doc_auto_cfg` feature, individual API elements don't need to be annotated. I believe we don't need to fork instability to get this working.

bugadani, 2024-11-21:
Apparently they also mark macro output https://github.com/ratatui/instability/blob/e365305716159aa89f1ce13d3e99e5e8c4438ad0/src/unstable.rs#L364

bugadani, 2024-11-21:
I was curious what it needs to tag feature-dependent API, and [this](https://github.com/probe-rs/rtt-target/pull/55/commits/553b355e295c2a4cba6eab3a33f89505660741e9) is all I had to do for rtt-target. Then, building docs with (sorry for PowerShell) `$Env:RUSTDOCFLAGS="--cfg docsrs"; cargo +nightly doc --all-features -p rtt-target` displayed the tags correctly.

PR #2628 (2024-12-02): Unstable documentation proof of concept
Closes #2498

[image]
closed 2024-12-02 by MabezDev