esp-rs/esp-hal #2500: RFC: Mark `Config` structs as `#[non_exaustive]` unless proven to be feature complete

MabezDev, 2024-11-08:
Whilst this is technically already part of the API-guidlines, I've seen some pushback in doing this we lose the ability to partially initialize a struct and finish the rest with `..Default::default()`.

We need to discuss alternatives, if there are any, and if not what we can do to ease this issue.

bjoernQ, 2024-11-20:
Only alternative to the ugly `{ let mut cfg = Config::default(); cfg.something = 42; cfg }` I can think of would be a builder - since configs are "easy" to build we probably could just use one of the derrive-builder crates to avoid writing that annoying code

I'd say then we could be less shy in future-proofing those structs 🤷‍♂️ 

Another thing: We probably want to mark enums retuned from the API as `#[non_exhaustive]`, too. Especially errors

jessebraham, 2024-11-25:
Will just mention the [Builder Lite](https://matklad.github.io/2022/05/29/builder-lite.html) pattern too, as it saves us having a bunch of extraneous structs.

bugadani, 2024-11-25:
A derive macro would help a lot here, the basic cases aren't very complex. Some exist on `crates.io` but all I found uses the full builder, not the light one.

jessebraham, 2024-11-25:
> A derive macro would help a lot here, the basic cases aren't very complex. Some exist on `crates.io` but all I found uses the full builder, not the light one.

I can look into rolling our own in our procmacros crate, good idea

MabezDev, 2024-11-29:
I don't think we need to do this for all drivers to close this. #2614 adds it for the drivers we plan to stabilize. I think to close this we should add a note about deriving this for configs in the API-GUIDELINES.

bjoernQ, 2024-11-29:
Maybe we should also add a builder-light for `esp_hal::Config`

PR #2686 (2024-12-05): Mention that config structs should derive `procmacros::BuilderLite` in the API guidelines
Closes #2500
closed 2024-12-05 by JurajSadel