esp-rs/esp-hal #2503: SPI Master additional features

Dominaezzz, 2024-11-08:
The SPI Master is currently not exposing some hardware features.
I'm listing them here to begin with rather than create a barrage of issues.

- [ ] Configurable Segmented Transfers (All chips except the base ESP32 support this. It allows to setup multiple transfers at once, rather then send them one by one with the CPU).
- [ ] Expose all the CS signals (The SPI hardware can take multiple CS pins and turn them on/off for each transfer. Software CS shouldn't be the long term solution for shared buses)
- [ ] Expose the `KEEP_CS_ACTIVE` option (This is important for chunked transfers in a single transaction)
- [ ] Allow splitting the SPI Buffer into two halves (All chips support doing this, and it allows for double buffering without DMA)
- [ ] Support async in CPU mode
- [ ] Expose SPI interrupts
- [ ] Expose SPI App interrupts (ESP32-S3 SPI supports these things called "app interrupts" which behave just like `SoftwareInterrupt`s, they are very handy for implementing an interrupt based queue)
- [ ] Expose the extra LCD features that the S2 has for the SPI hardware.
- [ ] Allow using the RX half and TX half in different (CPU/DMA) modes (Sometimes you only really need DMA in one half of the SPI transfer. This one might be difficult to support)
- [ ] Allow using SPI with `Dma[Tx|Rx]Buffer`s that don't have a length. The driver should accept a length in addition to the DMA buffer. (This allows for streaming transfers)

https://github.com/esp-rs/esp-hal/issues/2494 was created while I was drafting this issue. I think this issue should be separate since that one will be closed once the analysis is done, and this should remain open as a request for the features.

bugadani, 2024-11-08:
Please don't remove needs-attention, the team will do that during/after triage.

Dominaezzz, 2024-11-08:
Noted

bugadani, 2024-12-04:
C2/C3/C6/H2 also have APP interrupts.