esp-rs/esp-hal #2505: I2C should expose Ack settings for each transfer (like esp-idf does)

Dominaezzz, 2024-11-08:
## Motivations

The I2C hardware can be used to speak SCCB which is a protocol similar to I2C used to speak to cameras.
`esp-hal` currently abides by `embedded-hal::i2c`'s ack rules but those rules don't apply to all devices, like those that speak SCCB.

## Solution

Perhaps each transfer method (write, read, etc.) should take an extra config parameter to configure the ack settings.

## Alternatives

Write a 3rd party driver...

## Additional context

Not at the moment.

Dominaezzz, 2024-11-08:
Ah no, I definitely don't want direct SCCB support in the HAL.
I just want to be able to set these and do it myself.
- `ack_exp: Ack`
- `ack_check_en: bool`
- `ack_value: Ack`

Maybe SCCB could be a community crate. (Though the protocol isn't necessarily consistent across all cameras anyway, so maybe not worth doing, idk)

MabezDev, 2026-10-27:
Duplicate of #2598, which tracks a low-level I2C command interface that lets you configure acknowledge checking per transfer.

closed 2026-10-27 by MabezDev