esp-rs/esp-hal #2507: I2C: support 10-bit addressing

bugadani, 2024-11-08:
As noted by https://github.com/esp-rs/esp-hal/issues/2493#issuecomment-2464822742

We have a few options to support 10-bit addressing in inherent methods:

- Create separate `_10b` methods
- Change the address parameter into:
  - `enum Address {Seven(u8), Ten(u16)}` - too verbose?
  - `impl Into<Address>` and let the user pass either `u8` or `u16` - too implicit?

Any other, perhaps better ideas?

bugadani, 2024-11-08:
> There's some precedence for this in the I8080 driver

I'm just not entirely in love with the idea that users will need to add `_u8` or `_u16` at the end of their addresses, always. Though I guess we can implement `TryFrom<i32>` as well and pick one option (or "narrowest that fits") for a bare numeric literal.

bjoernQ, 2025-01-07:
In theory after #2864 this is not strictly a blocker since we _should_ be able to add 10-bit addressing support in a non-breaking way later

MabezDev, 2025-01-14:
> In theory after #2864 this is not strictly a blocker since we _should_ be able to add 10-bit addressing support in a non-breaking way later

I agree, I think that fact we can support this post 1.0 is enough now. 10bit support is rare enough that I don't think we _need_ this for 1.0.