esp-rs/esp-hal #2513: Cull examples

MabezDev, 2024-11-11:
The examples have grown beyond what is maintainable. Whilst longer term, we will want to move to a more "project" based list of examples, we can improve what we have here by culling useless ones.

I think if the following criteria is met, we can remove the example:

- The example is basic in nature, just initializing the driver and basic usage
- The driver has docs to explain the initialization/basic usage
- The driver has at least one test in `hil-test`

jessebraham, 2024-11-11:
I'll also add that if a driver _does not_ have adequate documentation to demonstrate initialization and use of the peripheral, but this is all the example is doing, we should just update the relevant documentation and then purge the example :)

Dominaezzz, 2024-11-11:
Is the thinking here that any new drivers introduced don't need to have examples, as long as they have docs describing basic usage and at least one HIL test, then it's fine to land? (I have a specific PR in mind and I'm trying to figure out if the example is basic or not)

MabezDev, 2024-11-13:
We discussed this a bit today, we should move "examples" that are actually manual tests (not something we can test in HIL, like displays) into their own project/directory.

> Is the thinking here that any new drivers introduced don't need to have examples, as long as they have docs describing basic usage and at least one HIL test, then it's fine to land? (I have a specific PR in mind and I'm trying to figure out if the example is basic or not)

If it's #2415, we'd probably need one of those manual tests for this (that we can run on the official espressif devkit).

playfulFence, 2024-11-14:
I'll start sorting through the examples a little by little. 
If anyone wants to take a part - let me know first so we can avoid working on the same thing 🤝

MabezDev, 2024-11-25:
Closed via #2547, we'll need to work more on examples in the future, but culling these useless ones has been a good start!

closed 2024-11-25 by MabezDev