esp-rs/esp-hal #2583: BLE examples replicate a UUID for both service and a characteristic

akauppi, 2024-11-21:
## Bug description

The BLE examples having the same UUID for both service and one characteristic seems like a mistake.

Check these lines:

- https://github.com/esp-rs/esp-hal/blob/f81b5f6c7f9292eb2687a60b1cc55bcf79ce5c84/examples/src/bin/wifi_ble.rs#L113
- https://github.com/esp-rs/esp-hal/blob/f81b5f6c7f9292eb2687a60b1cc55bcf79ce5c84/examples/src/bin/wifi_embassy_ble.rs#L143

```
        gatt!([service {
            uuid: "937312e0-2354-11eb-9f10-fbc30a62cf38",
            characteristics: [
                characteristic {
                    uuid: "937312e0-2354-11eb-9f10-fbc30a62cf38",
                    read: rf,
                    write: wf,
                },
```

This works just fine - I don't think the id's of the characteristics vs. their enclosing service really matter - but logically there shouldn't be any connection between the two?

The other characteristics are varied from the same theme, by changing one character. I would suggest to do the same here.

Background about the selection of UUIDs in [a StackOverflow answer](https://stackoverflow.com/a/21403170/14455).

## To Reproduce

Observe the links above.

## Expected behavior

The BLE samples not only are functional, but also show "intention", i.e. lead the reader in the right direction.

## Environment

Does not matter.

bjoernQ, 2024-11-22:
We plan to overhaul all the examples including the BLE ones. e.g. something mimicking a real-world device (e.g. a BLE temperature sensor or a HID keyboard)

akauppi, 2024-11-22:
Sounds great! I'll keep this ticket open until then, and remove once the new stuff is out. No rush.

akauppi, 2024-11-22:
One note on the examples - since the `gatt!` shows up in the code snippet. Rust Rover IDE has currently a problem with the `bleps` dependency, because it's not a published package (and uses features). Not your fault, but... since you plan to revise the examples, it's appreciated if outside deps would be ... less ad-hoc, I guess.

end of tangential :)

lure23, 2024-11-23:
For comparison, Nordic mentions this "LED Button Service" in their BLE tutorial: https://docs.nordicsemi.com/bundle/ncs-latest/page/nrf/libraries/bluetooth/services/lbs.html#lbs-readme

It has a service UUID and two characteristics, whose UUIDs are derived from that (differ by one char).

bjoernQ, 2025-08-29:
given we removed the old examples this can be closed