esp-rs/esp-hal #2584: Auto generate chip support status for docs

MabezDev, 2024-11-21:
We (mostly @jessebraham's work :D) used to manually maintain a manual table in https://github.com/esp-rs/esp-hal/issues/19, whilst this was mostly for tracking purposes, something like this is useful for new users to see the support status for the chip they wish to use at a glance.

I think we can use esp-metadata to generate a markdown table of chip support status for a given chip automagically.

jessebraham, 2025-03-19:
I have a local branch where I am able to generate a Markdown table using `esp-metadata`, however I'm not really sure how useful this is in practice. I guess based on the comment above what we really want is a table showing which _drivers_ are supported for each chip, rather than which peripherals are available, but it's unclear to me how to accomplish this exactly, as this information is not encoded in `esp-metadata` anywhere.

jessebraham, 2025-04-01:
@MabezDev sorry for the ping, but any feedback here?

MabezDev, 2025-04-04:
Looking a bit closer at esp-metadata's contents, you're right it's not that useful. What we can do it pivot slightly and use the toml-to-mdtable generating code you have with a manually curated toml file. The idea being that a toml file is easier to edit than the table itself. We can then add a PR checklist item to modify this toml file if the PR adds functionality. I think the toml file can live in the root of esp-hal package.

I imagine the toml file will be a map of `Driver -> [chip support list]`, something like 

```toml
[driver_support]
spi = ["esp32c3", "esp32"]
```

We might want a shorthand "all" entry for drivers that are supported on all chips? Not sure.

Dominaezzz, 2025-04-04:
> We might want a shorthand "all" entry for drivers that are supported on all chips? Not sure.

Might turn into busy-work when a new chip is added, as I imagine not all drivers would immediately work on the new chip without some work.

jessebraham, 2025-04-07:
I've written a simple subcommand to generate a Markdown table from a TOML file in the format suggested by @MabezDev. Remaining questions:

- Was this file intended to be checked into version control? If so, where will it live? Do we care about formatting the generated output, or do we just want to leave it as-is?
- Do we want to differentiate between a device not having a peripheral, and the peripheral not _currently_ being supported? If so, how?
- How do we want to go about populating the TOML file? Not all `esp-hal` modules are drivers, and not all peripherals map cleanly to single drivers, e.g. do we want to specify `ADC`/`DAC` separately, or just have an `Analog` driver?

MabezDev, 2025-04-07:
* Yes, checked in so it can be updated with relevant PRs
* I haven't seen the formatting, but I don't think we need anything special here, just a table
* Yes, we do. I don't mind what indicators we use
* Manually, and in the case if Analog, I would expect ADC and DAC to show up separately

PR #3617 (2025-06-10): Peripheral support table
- I'm sure I've missed a peripheral or two somewhere
- The concept is a bit backwards, we probably will want to generate the peripheral instance cfgs (e.g. (`spi2`) based on the driver config. Right now the symbols are instead used to somewhat double-check the configs. This will be tidied up later.

Closes #2584