esp-rs/esp-hal #2596: Create non-trivial example applications which encapsulate all required dependencies and configuration

jessebraham, 2024-11-25:
This has been discussed internally, but I would like to open an issue to track progress on this and give a bit more visibility to the community regarding our plans here.

Our previous approach to examples was to create various applications per-peripheral, demonstrating their use. This resulted in a large number of examples, many of which were not particularly useful, and was a burden from a maintenance standpoint. Additionally, many of these examples were in reality more test-like in nature than they were useful examples. Many of these have been removed now, and converted to doctests where able.

We would like to create a number of non-trivial examples, each of which standalone projects. These should ideally demonstrate use of multiple peripherals in more real-world applications. We can take inspiration from ESP-IDF, for example. These applications should encapsulate all dependencies and configuration required to build them. This would alleviate the issues with our giant `Cargo.toml` containing dozens of dependencies, and hopefully make it a bit easier for users to get started. This will hopefully also help reduce the amount of duplication using our previous/current approach.

We should spend some time brainstorming what we would like these examples to look like. One idea would be something like reading one or more sensors, and sending the readings over the network to an MQTT broker, or something of this nature. Input is welcome (and encouraged) here.

Some relevant issues:

- https://github.com/esp-rs/esp-hal/issues/1436
- https://github.com/esp-rs/esp-hal/issues/2513
- https://github.com/esp-rs/esp-hal/issues/1432

MabezDev, 2024-11-28:
Just posting as an FYI, for some more interesting examples we can use [m5stack](https://m5stack.com/) devkits, as well as the official Espressif ones as they are a partner of ours.

Dominaezzz, 2024-11-29:
If M5Stack is on the table I'm willing to contribute an example of an HTTP MJPEG Server using the [Unit CamS3 5MP](https://docs.m5stack.com/en/unit/Unit-CAMS3%205MP). It's SCCB/I2C API is very simple to work with compared to other cameras.

This example would use LCD_CAM, I2C and WiFi. (In theory one could make use of the microphone, led and SD card but that's out of scope for me).

bjoernQ, 2024-12-04:
Some topics I'd like to see covered by the new examples (in no particular order and incomplete) based on questions and discussions:

- I2C + SPI displays via embedded-graphics
- MQTT
- TLS
- Provisioning (SoftAP / BLE?) - includes persisting data
- OTA
- Sleep (deep + light)
- I2S (e.g. a simple MP3 radio player?)
- BLE HID keyboard
- maybe using a REST API

Ideally as multi-target projects (if applicable)

bjoernQ, 2024-12-04:
Maybe when combined with good documentation the examples might replace https://github.com/esp-rs/no_std-training - since it's quite unpleasant to maintain (thanks to @SergioGasquez for always taking care of that btw)

playfulFence, 2024-12-10:
Just in case, I started trying to convert the `esp-clock` example from Wokwi to the current version of our bare-metal drivers.

JurajSadel, 2024-12-11:
Regarding MQTT we can use our example from DevCon23: https://github.com/JurajSadel/esp32c3-no-std-async-mqtt-demo. We have one for C3, ESP32, and S3 at the moment. All of them need to be updated to latest hal - I will update them.

bjoernQ, 2025-04-03:
See https://github.com/esp-rs/esp-hal/pull/1899#issuecomment-2765280523

Users will sooner or later need TLS - we have a few options: embedded-tls, esp-mbedtls and Rustls

Providing examples or some general hints would be useful

bugadani, 2025-04-03:
We could maybe implement a firmware update demo, that uses embassy-usb, acts as USB mass storage, and flashes a binary using esp-storage and the esp-bootloader-esp-idf crates.

David-OConnor, 2025-05-18:
Here's an idea for a quick addition relevant to a case I ran into recently: Simple program that configures a timer that auto-reloads, triggers an interrupt, and starts. Configures UART with a few non-default settings. Store the timer and UART structs to global state. In the Timer ISR, clear the interrupt, and send a message over UART.

Bonus 1: Set up Uart for a receive interrupt; in its ISR, read the message.

Bonus 2: Do the UART using DMA.

Some notes: There appears to be no obvious way to configure the Uart, as its config struct's fields (e.g. for BAUD) are private, and the method to set baud is also private. Easy patch: Make the fields public, and let you override the non-defaults with a struct literal. Or add a `set_config` method etc. (Or maybe I'm missing something?) Another note: The `Uart` seems to be non-trivial to borrow when in a Mutex. (Experiencing a strange issue. I suspect this is related to its <`static, ...> trait requirement, which I haven't seen used before for this application.

jessebraham, 2025-09-16:
After some internal discussion, I am going to go ahead and close this issue.

The example restructuring in #3915 alleviates some of the pain points which led to the creation of this issue, and instead we will create a more interesting project replacing our current `no_std` training book. Please refer to the RFC in the training repository for further updates: https://github.com/esp-rs/no_std-training/issues/80