esp-rs/esp-hal #2604: I2C: allow disabling bus arbitration (? esp-idf doesn't support it)

bugadani, 2024-11-25:


jessebraham, 2024-12-19:
I believe this is simply a matter of setting `I2C_ARBITRATION_EN` in `I2C_CTR_REG`, so can likely be a configuration value.

bjoernQ, 2024-12-19:
Seems like that

it's defined with a RST-value of 1 but we do `self.register_block().ctr().write` and don't set it - so we apparently always disable it

bugadani, 2025-01-20:
Actually, the `w` in write is initially set to the reset value, which is in the case of the C3 (just the one I'm looking at) is 0x20b, so I think we should have I2C_ARBITRATION_EN always on.

bjoernQ, 2025-01-20:
👍 didn't thought `write` takes the RST-value into account and even more didn't realize we correctly define the RST-value

but still agree on @jessebraham comment - should be possible to add it in a non-breaking way later

(and seems ESP32 doesn't have that flag?)