esp-rs/esp-hal #2608: I2C: allow configuring pin filters

bugadani, 2024-11-25:


bjoernQ, 2024-12-19:
I guess the issue refers to this from the TRM
```text
28.4.2 SCL and SDA Noise Filtering

SCL_Filter and SDA_Filter modules are identical and are used to filter signal noise on SCL and SDA,
respectively. These filters can be enabled or disabled by configuring I2C_SCL_FILTER_EN and
I2C_SDA_FILTER_EN.

Take SCL_Filter as an example. When enabled, SCL_Filter samples input signals on the SCL line continuously.
These input signals are valid only if they remain unchanged for consecutive I2C_SCL_FILTER_THRES
I2C_SCLK clock cycles. Given that only valid input signals can pass through the filter, SCL_Filter can remove
glitches whose pulse width is shorter than I2C_SCL_FILTER_THRES I2C_SCLK clock cycles, while SDA_Filter
can remove glitches whose pulse width is shorter than I2C_SDA_FILTER_THRES I2C_SCLK clock
cycles.
```

Sounds like it's just something we can add to `Config`

jessebraham, 2025-01-17:
This should be able to be done in a non-breaking way in that case, so I've updated the label accordingly. If this is incorrect then feel free to change it back.